Ultimaker/Cura #6978: Version 4.4.0 and version 4.4.1 ignore nozzle diameter changes.

ReHoAeb, 2020-01-15:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

ReHoAeb, 2020-01-15:
Version 4.4.0 and version 4.4.1 ignore nozzle diameter changes.

Twigho, 2020-01-15:
You need to use the comma. (0,4 instead of 0.4 ) :-/

mahtDFR, 2020-01-15:
Thanks for copying the issue template but it doesn't appear to be populated with info. Can you tell me more about your printer setup so I can try to replicate the issue?

Amapem, 2020-01-15:
Application version
4.4.1

Platform
Windows 10 64bits

Printer
Artillery Sidewinder X1 prile here : https://3d-nexus.com/resources/file-archives/category/8-artillery-evnovo

Reproduction steps
Every thing is OK with Cura 4.2.1 - see screenshot :
[image]

and with Cura 4.4.1 : nozzle size is empty  and there is no quality profile also - see screenshot :
[image]

Log file
[cura.log](https://github.com/Ultimaker/Cura/files/4067586/cura.log)

mahtDFR, 2020-01-17:
This issue seems to be related to broken nozzle size definitions.

Version 4.4 introduced an extra layer of the settings stack for Intent profiles. As a result your nozzle size metadata (which is configured correctly for 4.2 but not 4.4) is getting ignored. Read the discussion here for more detail: #6666  

A workaround would be to edit the metadata and populate it with the correct info. This is located in the resources folder of the install folder. Also found some clues in the log you provided:

`2020-01-14 22:18:38,411 - ERROR - [MainThread] UM.Logger.logException [88]: UM.Settings.ContainerFormatError.ContainerFormatError: When trying to deserialize artillery_base_extruder_0 #2, we received an unknown container ID (artillery_sidewinder_x1_0.4)`

You could also nudge the developer of this profile to update it for compatibility with 4.4+ They could even make a pull request to merge their printer profile directly with our repo so it can be included it in the printer selection menu by default - the result of that is a more user-friendly experience with less messing around with the file system by the user.

Hope this gives some clues

Vandrasc, 2020-01-17:
@ReHoAeb , did you use the comma as mentioned by @Twigho ?

@Amapem, is that after an upgrade from 4.2 to 4.4? If so, can you try w/o an upgrade?

Twigho, 2020-01-17:
Maybe you can try changing this Windows setting: (A very old tip was to replace the comma with the dot) 
[image]

Amapem, 2020-01-17:
> @Amapem, is that after an upgrade from 4.2 to 4.4? If so, can you try w/o an upgrade?

This was with a new cura 4.4.1 installation.
I didn't see how to upgrade cura, I've both versions in different folders.

mahtDFR, 2020-01-17:
> Maybe you can try changing this Windows setting: (A very old tip was to replace the comma with the dot)
> [image]

Just tried this but it didn't work for me unfortunately.

> > @Amapem, is that after an upgrade from 4.2 to 4.4? If so, can you try w/o an upgrade?
> 
> This was with a new cura 4.4.1 installation.
> I didn't see how to upgrade cura, I've both versions in different folders.

You can install a later version of Cura without overwriting an old installation, just be sure not to remove the old version using the installer. The application and the appdata will be stored in separate folders

Amapem, 2020-01-18:
> This issue seems to be related to broken nozzle size definitions.
>
> Version 4.4 introduced an extra layer of the settings stack for Intent profiles. As a result your nozzle size metadata (which is [...]

Can you explain the Metadata to edit?

mahtDFR, 2020-01-20:
After some discussion we have decided to defer this bug. Given that this is a third party profile that's not in our repo, and we don't have a printer here for testing, we can't devote our resources to it.

Twigho, 2020-01-20:
to find your nozzle, it must change the line "setting_version = 8" with "setting_version = 10" on the files in the folder "variants". 
For the quality and the material, I believe that it make a file for each material and each quality, in the folder "quality". (Look at the files already present for an example. (It sounds very boring))

mahtDFR, 2020-01-20:
Confirmed. Updating the setting version from 8 to 10 in the variants makes it register the nozzle size in 4.4.1
Tested for generic PLA using 0.4 nozzle.
Well done @Twigho !
[image]

Twigho, 2020-01-20:
To retrieve the profiles: 
set the line  ' "has_machine_quality": true, ' to false in the file artillery_base.def.json in the folder definition. All seem unlock. 
[image]

mahtDFR, 2020-01-21:
Will close this off now as it seems solved.

closed 2020-01-21 by mahtDFR

Amapem, 2020-01-21:
Another post : 
https://www.thingiverse.com/thing:4117423