Ultimaker/Cura #6979: Issue importing multiples .stl files

leoheck, 2020-01-15:
Using Cura 4.5 (from a daily Ubuntu PPA)
```
deb http://ppa.launchpad.net/thopiekar/cura-master/ubuntu eoan main
```
I am fixing this manually every upgrade I made so it is better to report it here.
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 1019, in event
    return super().event(event)
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 442, in event
    event._function_event.call()
  File "/usr/lib/python3/dist-packages/UM/Event.py", line 189, in call
    self._function(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
    func(dest, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 1727, in _readMeshFinished
    arranger = Arrange.create(x = machine_width, y = machine_depth, fixed_nodes = fixed_nodes)
  File "/usr/lib/python3/dist-packages/cura/Arranging/Arrange.py", line 72, in create
    if not points:
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
```

I am fixing it by replacing line 72 by line 73
[image]

leoheck, 2020-01-15:
Cura logs if needed.

```
Thread 0x00007fe7d37fe700 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "/usr/lib/python3.7/threading.py", line 870 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe7d3fff700 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.7/threading.py", line 870 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe7f17fa700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1794 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1365 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe7f1ffb700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "/usr/lib/python3.7/threading.py", line 870 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe7f27fc700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1794 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1219 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe7f2ffd700 (most recent call first):
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1140 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe7f37fe700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/home/lheck/.local/share/cura/4.5/plugins/OctoPrintPlugin/OctoPrintPlugin/zeroconf.py", line 1583 in wait
  File "/home/lheck/.local/share/cura/4.5/plugins/OctoPrintPlugin/OctoPrintPlugin/zeroconf.py", line 1167 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe7f3fff700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/home/lheck/.local/share/cura/4.5/plugins/OctoPrintPlugin/OctoPrintPlugin/zeroconf.py", line 1583 in wait
  File "/home/lheck/.local/share/cura/4.5/plugins/OctoPrintPlugin/OctoPrintPlugin/zeroconf.py", line 1031 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe800c92700 (most recent call first):
  File "/home/lheck/.local/share/cura/4.5/plugins/OctoPrintPlugin/OctoPrintPlugin/zeroconf.py", line 960 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe824ff9700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Backend/Backend.py", line 150 in _storeStderrToLogThread
  File "/usr/lib/python3.7/threading.py", line 870 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe8257fa700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Backend/Backend.py", line 138 in _storeOutputToLogThread
  File "/usr/lib/python3.7/threading.py", line 870 in run
[... 82 lines cut ...]

Thread 0x00007fe84bfff700 (most recent call first):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 483 in read
  File "/usr/lib/cura/plugins/USBPrinting/AutoDetectBaudJob.py", line 74 in run
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 121 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fe8609d7700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/threading.py", line 427 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Current thread 0x00007fe88a3b5740 (most recent call first):
  File "/usr/lib/python3/dist-packages/cura/CrashHandler.py", line 363 in _logInfoWidget
  File "/usr/lib/python3/dist-packages/cura/CrashHandler.py", line 163 in _createDialog
  File "/usr/lib/python3/dist-packages/cura/CrashHandler.py", line 84 in __init__
  File "/usr/bin/cura.real", line 136 in exceptHook
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 356 in exec_
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 819 in run
  File "/usr/bin/cura.real", line 205 in <module>
```

Ghostkeeper, 2020-01-15:
Duplicate of https://github.com/Ultimaker/Cura/issues/6972.

closed 2020-01-15 by Ghostkeeper