Ultimaker/Cura #6981: Print one at a time Error

BixlerVDL, 2020-01-15:
**Application version**
Cura 4.4.1

**Platform**
Windows 10

**Printer**
Creality CR10s

**Reproduction steps**
1. Selected print one at the time, print sequence
2. Loaded two stl files like usual, both small enough to print one at the time

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Build volume had been reduced almost all the way down, Objects were unable to be placed right next to each other. The objects show a ring around them, which is correct, the rings cant overlap because otherwise the carriage or the metal frame would hit the object whilst printing. But even though they did not overlap Cura was not able to slice successfully.

I reported this issue before, I saw that their was a fix in the bug fix section, but I am pretty sure that nothing changed. Main error is that the build volume is reduced by way too much.

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-01-15:
> But even though they did not overlap Cura was not able to slice successfully.

Can you elaborate? What is happening in Cura and what would you expect to happen? Do you have a log file for us? Do you have a project file for us?

BixlerVDL, 2020-01-15:
So I want to print multiple models, but I want them to print individually one after the other. I go to the print sequence section in cura settings, and then I select print one at the time. I believe that Cura should know print the models on the build plate one by one, instead of layer by layer.

Ghostkeeper, 2020-01-15:
Yeah, I believe it does. When I load two cubes and set the sequence to One At A Time, for me they get printed one by one, not layer by layer.

BixlerVDL, 2020-01-15:
[image]
[image]
[image]

BixlerVDL, 2020-01-15:
Gantry height is fine, at least it should be if the models are separated so much.

Ghostkeeper, 2020-01-15:
Yeah, looks like Cura is saying that the models don't fit the build volume. Again, do you have a log file and a project file for us so that we can reproduce the problem? I'm loading two cubes in that are clearly lower than the gantry height and it slices fine.

BixlerVDL, 2020-01-15:
Let me get it...

BixlerVDL, 2020-01-15:
But still, if the two models are taller than the gantry height, it should still be possible to print the models. As long as the models are not on the path of the gantry.

BixlerVDL, 2020-01-15:
Where can I get the log files?

Ghostkeeper, 2020-01-15:
> But still, if the two models are taller than the gantry height, it should still be possible to print the models. As long as the models are not on the path of the gantry.

That would depend on the shape of the gantry. The shape of the gantry is not modelled in Cura at the moment, because that would require forward kinematics which is really expensive to compute (especially in Python).

> Where can I get the log files?

You posted a link to it in your original post.

BixlerVDL, 2020-01-15:
Can you tell me how I can find the log files?

Ghostkeeper, 2020-01-15:
Yes, the link you posted explains it, depending on your operating system: https://github.com/Ultimaker/Cura#logging-issues

BixlerVDL, 2020-01-15:
I tried it, 
C:\Users\Van_lu>/AppData\Roaming\cura\4.4.1\cura.log
The system cannot find the path specified.

BixlerVDL, 2020-01-15:
Cura does slice if I increase the gantry height from 25mm to 100mm. I will print the models to see if that solved the issue...

Ghostkeeper, 2020-01-16:
> I tried it,
C:\Users\Van_lu>/AppData\Roaming\cura\4.4.1\cura.log
The system cannot find the path specified.

That would be `C:\Users\Van_lu\AppData\Roaming\cura\4.4\cura.log` (if your username is `Van_lu`).

BixlerVDL, 2020-01-16:
Okay, i got it!
[OneAtTheTimeError.log](https://github.com/Ultimaker/Cura/files/4070282/OneAtTheTimeError.log)

mahtDFR, 2020-01-16:
It seems the gantry height is lower than your models. We can't model the gantry height for reasons explained above.

closed 2020-01-16 by mahtDFR

BixlerVDL, 2020-01-16:
I understand, I guess the only way to overcome this issue is by increasing the gantry height and manually checking if the objects will collide with the print head. 

I am a bit surprised though, with the technology and software available nowadays I am sure that an improvement to the software would allow for better one at the time printing. 
I did see that in the bugfix section it said that there was a fix for one at a time printing, what exactly was the fix?

Thank you for helping me find the cause of the issue!

Ghostkeeper, 2020-01-17:
> I am a bit surprised though, with the technology and software available nowadays I am sure that an improvement to the software would allow for better one at the time printing.

Improvements are possible, certainly with the last model being allowed to be higher than the gantry (there are other issues about this open). The general problem of the gantry having a certain shape is also theoretically possible but computationally expensive to do, and will only work if someone enters the shape of the gantry correctly. So we've decided on a "shape" of an infinite plane which is on the safe side.

> I did see that in the bugfix section it said that there was a fix for one at a time printing, what exactly was the fix?

For 4.4 we've made fixes to the ordering of models in one-at-a-time mode, as it was apparently not [taking the adhesion method's area into account properly](https://github.com/Ultimaker/Cura/commit/95120300601c5a8391a3909c7291f95248d35052). For 4.5 we have a small fix ready for a bug that limited the number of vertices in the print head shape.