Ultimaker/Cura #6982: Any chance of a configurable volumetric speed limit?

x0rtrunks, 2020-01-15:
**Is your feature request related to a problem? Please describe.**
It's currently possible to use a print speed/layer height/extrusion width combination that exceeds the volumetric extrusion limit of the hotend. To avoid it, you must know the limit, and do some math to ensure your settings don't exceed it. Exceeding the limit can lead to under extrusion, stripping filament, poor surface finish, etc. 

Volumetric flow defined as **Max Print Speed * Extrusion Width * Layer Height**

I would think this would be useful for variable layer height printing as it could scale the print speed on the taller layers to remain within the extrusion limit without unnecessarily slowing down all the layers.

Your hotend has a melt rate limit, it can only push so much plastic reliably. For example, 10mm/s^3. Would it be possible to have a configurable value that would allow you to limit/warn when your speed and layer height/extrusion width combination exceeds the configured value.

The limit can be found through testing, by extruding to free air at increasing speed until the extruder starts to skip. Then taking the feed rate where skipping happens and the filament diameter and you can calculate the volume in mm^3/s with **Max Flowrate = Max Input Feedrate * pi * (Filament Diameter/2)^2**

**Describe the solution you'd like**
I would like a parameter that takes the volumetric extrusion limit I give based on the test of my extrusion system, and have that value either limit the print speed in order to stay within the volumetric limit, or at least highlight/warn that my combination could result in under extrusion, etc. A check box for on or off, and a field to input the limit in mm^3/s

**Describe alternatives you've considered**
If limiting the print speed automatically isn't desirable, at least a warning could give the user a heads up, possibly with a suggested max speed given the current layer height and extrusion width.

**Affected users and/or printers**
Any printer and any user could benefit from this information.

**Additional context**
Slic3r has a volumetric extrusion limit function, but it doesn't really provide any warning or feed back to the user when the limit is being applied, but it does work nicely for preventing accidental under extrusion.

Thanks

Ghostkeeper, 2020-01-16:
Duplicate of https://github.com/Ultimaker/Cura/issues/5248.

closed 2020-01-16 by Ghostkeeper

rcmonitor, 2021-02-14:
+1 for mm/time unit setting (mm/s or mm/min).
Volume of material may vary (the filament spool may be of let's say 2.90 instead of 2.85, or 1.70 instead of 1.75), but the performance of extruder on filament may not.