Ultimaker/Cura #6984: Material Prefrences Dont Import to 2nd Printer

PeterGarthAmos, 2020-01-16:
Hi, when creating a new material under my first printer (prusa i3) everything works (apart from haveing to exit and open to see the changes). I then export the materia profile. So far so good. Then i change my printer to the second printer (tevo black widow) and there are no material profiles to choose from. I then select import and select the new profile that I exported from my first printer. It says import successful but there is nothing visible in the materials. However when going back to the first printer the material has imported back into  that printer. (there is now a duplicate of the profile.)
Seems like I cannot import any materials into my second printer profile, even when operating from that profile the imports go to the first printers  profile.

mahtDFR, 2020-01-16:
It's possible that something is being overridden. It would be helpful to have a project file (3mf) or log output to decipher this

PeterGarthAmos, 2020-01-17:
Thanks for the comment, how can I provide one of those?

PeterGarthAmos, 2020-01-17:
If you add a new printer (the Tevo Black widow in my case) you cant even create a new material. The create button is there but nothing happens. I've uninstalled it and re installed it and the program but no joy. However everything works under the first printer (prusa i3)

mahtDFR, 2020-01-17:
When you click post issue it gives you a template. Pasted below for you. Please fill it out as fully as you can and include logs/project files where possible.

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

PeterGarthAmos, 2020-01-17:
**Application version**
Cura 4.4.1

**Platform**
Windows 10

**Printer**
two printers, 
1. Prusa i3 (this on was installed first and the materials are all there and working fine.)
2. Tevo black widow. (when selecting this the materials section has no materials to select other than the two headings favirots and generic. next to those are the materials as unknown.)
**Reproduction steps**
when creating a new material under my first printer (prusa i3) everything works (apart from haveing to exit and open to see the changes). I then export the materia profile. So far so good. Then i change my printer to the second printer (tevo black widow) and there are no material profiles to choose from. I then select import and select the new profile that I exported from my first printer. It says import successful but there is nothing visible in the materials. However when going back to the first printer the material has imported back into that printer. (there is now a duplicate of the profile.)
Seems like I cannot import any materials into my second printer profile, even when operating from that profile the imports go to the first printers profile.

If you add a new printer (the Tevo Black widow in my case) you cant even create a new material. The create button is there but nothing happens. I've uninstalled it and re installed it and the program but no joy. However everything works under the first printer (prusa i3)

**Steps taken**
I have reinstall cura. reselected the printers with the same results.
Tried just having the tevo black widow printer installed. Same results. 

**Screenshot(s)**
[image]
[image]

**Actual results**
Still not working

**Expected results**
I should be able to select and create materials for the other printer.

Project file
NA
Log file
NA

I cant see that I can give anymore detail than this. If you select and install the tevo black widow from curas drop down menu are you able to select different materials?

PeterGarthAmos, 2020-01-17:
I have tried adding one or two other printers and the material section seems to work fine. Seems to be specific with the Tevo Black Widow selection.

PeterGarthAmos, 2020-01-17:
so tried it on two different computers with the same results. the Tevo Black widow does not have the selection of materials.

Ghostkeeper, 2020-01-17:
The Tevo Black Widow printer definition doesn't support material profiles at all, so no available profiles should be listed there at all.

Should material profiles be available for the Tevo Black Widow? That's not for us Cura developers to decide, really. We depend on the Tevo community to maintain their profiles and right now they're basically saying that they don't have the resources to test for different materials and you should just always use their settings.

thebreadboard, 2020-01-18:
That explains why there are none by default, but not why We can not create our own, how do we fix this so we can create our own material profiles ?

thebreadboard, 2020-01-18:
Simple ish fix for TBW is to change the definition file here "C:\Program Files\Ultimaker Cura 4.4\resources\definitions\tevo_blackwidow.def.json" and change the setting 
**"has_materials": false,** to **"has_materials": true,**
DO THIS BEFORE ADDING THE TBW TO YOUR CONFIGURATION OR IT WILL CRASH CURA

the path to the file may differ on your install, but i'm guessing the definition file name will not

PeterGarthAmos, 2020-01-19:
thebreadboard
Thanks your fix worked like a charm.

closed 2020-01-19 by PeterGarthAmos

Ghostkeeper, 2020-01-21:
> That explains why there are none by default, but not why We can not create our own, how do we fix this so we can create our own material profiles ?

They basically say "no you're not allowed to create your own." So it's not allowed. You can of course still save the settings you modified to a normal profile. Or, like you found out yourself, you can mod the profile to allow it explicitly.

PeterGarthAmos, 2020-01-22:
GhostKeeper
Thanks guys. The fact that this is something that can me modified is great and It seems an easy fix. However most profiles are close, you think it would just be ok to allow the generic ones on all of your available printers. How bad could that be?

Ghostkeeper, 2020-01-24:
Well for instance, the Ultimaker 2 doesn't use the temperatures from your g-code nor the retraction length and speed. Only the fan speed will be set correctly. So you'd be very confused that the material profiles *simply don't work on the printer*.

PeterGarthAmos, 2020-01-31:
Nice to know. Thanks makes sense