Ultimaker/Cura #6991: Wall count change at Layer-range

FlailAway, 2020-01-16:
**Is your feature request related to a problem? Please describe.**
Nope

**Describe the solution you'd like**
Wall count or wall-thickness to be changed at Layer-range.

I am printing 70mm cube with an "L" shape wall on top at the outside edges of 2-sides. The wall is 4.5mm (inwards) thick and 20mm high with several horizontal holes. I need to have "wall line count"  = 5 for the holes in the walls so they do not crush up when a screw is tightened.

The cube is more than robust enough with 2-walls and 15% infill but for now I have to print it all with 5-wall-line-count. I could probably use a Mesh-fix for the holes but it needs to be done on each version of the cube+wall model and there are about 30 different models to Mesh.

Ideally something along the lines of the Post Processing, "Change at Z" but for walls. This would save me a bunch of time and lot of PLA. :) 

The extra walls could could sit on top of a flat surface or maybe transitioned up from the lower wall count if no flat to to build on.

Thanks for looking.

Ghostkeeper, 2020-01-17:
It's not feasible to do this in the Change at Z script. You could remove walls, but you'd also have to adjust the adjacent lines to make them connect to the next wall, and sometimes add skin lines and such.

This is part of a general story to enable some settings to be changed per layer, but that's not going to happen for the foreseeable future. This is a long-term idea.

3781637, 2021-04-20:
[image]
I've been trying to accomplish this using the Erasure block settings (whatever naming that should be), but the wall line count doesn't seem to work at all this way.

Ghostkeeper, 2021-05-06:
This is a project file demoing how an infill mesh can add more walls in certain places, a workaround for this issue: 
[infill_mesh_increasing_wall_count.zip](https://github.com/Ultimaker/Cura/files/6438146/infill_mesh_increasing_wall_count.zip)

closed 2023-07-13 by github-actions[bot]