Ultimaker/Cura #6993: Reduce or limiting the size of cura log files

DivingDuck, 2020-01-17:
**Is your feature request related to a problem? Please describe.**
This is a kindly request to limit the size of Cura log files. There is a discussion in the forum about this request. Please see https://community.ultimaker.com/topic/30873-reduce-or-limit-the-size-of-the-curalog-file/
The request was originally ask from Cuq for cura.log file and I like to support this too.
 
**Describe the solution you'd like**
Implementing a rolling log file with a user definable fix size would be a appropriate solution to fix the unlimited consumption of space. 
Maybe a simple future add-on for Cura development team to solve this in general for all log files.

**Describe alternatives you've considered**
NA, well maybe doing it old school deleting manually - in 2020?

**Affected users and/or printers**
It is good practice to to play with resources nicely instead of consuming only, so it will help everyone :)

**Additional context**
Looking forward to see this some when happen in the future as this is a low hanging fruit.

Best regards,
DD

nallath, 2020-01-20:
This will be in the 4.5 release. It's one of those things that was so simple that you wonder why it took you so long to build it (So thanks for pointing this out!)

closed 2020-01-20 by nallath

DivingDuck, 2020-01-20:
Great, this was fast. I appreciate this, thanks a lot.

mahtDFR, 2020-02-17:
For reference: CURA-7123
Will close this off now.

DivingDuck, 2020-02-18:
Thanks and yes, it seems to play nice now with Cura 4.5 Beta.