Ultimaker/Cura #6995: [4.4.1] On two color print with extruders 2 and 3, prime tower purges with extruder 1 before purging with correct extruder

PEstes, 2020-01-17:
**Application version**
4.4.1

**Platform**
Windows 10 64-bit, Intel i5, Intel UHD 620

**Printer**
Custom printer definition for Geeetech A20T (3 extruders, 1 nozzle). All nozzle offsets are 0.

**Reproduction steps**
1. Printing 2 color STL model by opening 2 different partial models
2. Assign extruder #2 to one STL model
3. Assign extruder #3 to other STL model
4. Merge
5. Enable prime tower, 35mm, 80mm3, same values for all 3 extruders
6. Set Build Plate Adhesion Extruder to #2
7. Slice

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Every time there is an extruder change, the current material is purged out by extruder #1. Only after doing so does it purge with the correct nozzle. You can see in the picture that the prime tower has black in it. This is from the #1 extruder, yet the #1 extruder is not assigned to any part of this print. There's basically two purges happening (ie #2 -> #1 -> #3 instead of just #2 -> #3) on every extruder change when only 1 should occur.

[image]

**Expected results**
Extruder #1 is not used in this print. Therefore, it should never be part of the prime tower. It should never be purged. It is simply wasting material unnecessarily as I need to increase the tower size/volume to accomodate the extra purge.

**Project file**

[makerbot_coin_mk3_m_a.zip](https://github.com/Ultimaker/Cura/files/4078147/makerbot_coin_mk3_m_a.zip)

**Log file**
N/A

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-01-20:
Getting an engine crash on this project file currently, which is a different issue in itself...

`2020-01-20 17:37:48,468 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [829]: Backend quit with return code -11. Resetting process and socket.`

Seems to be a segfault.

Ghostkeeper, 2020-01-20:
Can you try enabling support, setting the support extruder to 2, then disabling support again? Maybe that's a workaround for you.

PEstes, 2020-01-28:
I've tried changing the support extruder to extruder #2 and get the same result. Also tried setting it to extruder #3 and also same result as well.

Block137, 2020-01-29:
I remember reading it always prime with T0 because T0 is default to have highest "Adhesion Tendency", a permanently invisible setting in material tab, to avoid problem caused by #6909 

If Adhesion Tendency is accessible by user then this maybe configurable?

Ghostkeeper, 2020-01-30:
This makes sense actually. It seems that the prime tower code just takes all extruders and orders them from high Adhesion Tendency to low (and in case of ties the lowest extruder number wins).

https://github.com/Ultimaker/CuraEngine/blob/e904e260716d072d6c766dc73fbc18f7fb224fce/src/PrimeTower.cpp#L33-L38

It should only take the *used* extruders here, I'd think. So that's a bug then.

vwpollen, 2020-06-26:
Any chance for this issue to be fix in future revision ?

Ghostkeeper, 2020-07-02:
I hope so. We don't have a fix yet. The outlines for the prime tower are generated before it's known which extruders are being used, at the moment. Bit of a weird architecture choice that needs a small refactor.

smartavionics, 2020-07-02:
Hello @vwpollen, I don't know if it will help but my cura builds feature a "compact prime tower" setting that behaves differently inasmuch that it only primes the extruders used on each layer. 

For the project above (CFFFP_cube.3mf) I get the following output for extruders 2, 3 and 4. Extruder 1 isn't used at all.

[image]

[image]

[image]

If you want to try my builds, they can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there before using.

smartavionics, 2020-07-02:
I changed the material colours to make it more obvious, the prime tower has alternate yellow and green layers.

[image]

vwpollen, 2020-07-06:
@smartavionics, I have given it a try and this option fixes the issue. Do you have more info on this Cura branch name? Perhaps it can helps integrating this option in future "official" Cura revision ?

smartavionics, 2020-07-06:
Hi @vwpollen , I'm glad that this has helped you. I am now only contributing bug fixes to UM so this feature (compact prime tower) is not going to be in the official version.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

Block137, 2024-11-22:
No problem in 5.8.1 for me
[image]

closed 2024-12-07 by github-actions[bot]