Ultimaker/Cura #6996: Separate extruder for skirt and brim

Liger0, 2020-01-18:
**Is your feature request related to a problem? Please describe.**
While a skirt can use the same material as the piece to avoid waste of time or pricies soluble supports, brim can benefit of soluble materials so you don't have to post process it.
The extruder used is shared between skirt and brim.

**Describe the solution you'd like**
Use separate options for skirt and brim extruder material

**Describe alternatives you've considered**
Make the adhesion options all separate, so you can use skirt/brim and raft at the same time to purge and make the raft clean, and you can easily configure everything.

**Affected users and/or printers**
Multiple extrusion printers

Ghostkeeper, 2020-01-21:
This would require allowing skirt, brim and raft to be printed in the same print in the first place, of course.

TCPILOT, 2020-01-24:
Is this why I can't print a brim with my second extruder? It makes no sense to allow the selection of the second extruder if it will not print the brim.

Ghostkeeper, 2020-01-27:
Currently Cura still allows you to print the brim with a second extruder. It just doesn't allow you to automatically switch the extruder based on whether "skirt" or "brim" is selected.

GregValiant, 2023-09-15:
There currently are:
{skirt_brim_extruder_nr}
{adhesion_extruder_nr}
{raft_base_extruder_nr}
{raft_interface_extruder_nr}
{raft_surface_extruder_nr}
Is that sufficient?  Can we close this?

closed 2023-09-30 by github-actions[bot]