Ultimaker/Cura #6997: [4.4.1] Purge tower coordinates don't take into account hotend offset

Liger0, 2020-01-18:
**Application version**
4.4.1

**Platform**
Win10 x64

**Printer**
Custom dual extrusion

**Reproduction steps**
1. Second hotend +20mm offset
2. Enable purge tower
3. Try to slice

**Screenshot(s)**
[image]

**Actual results**
The purge tower is calculated on a place that the primary hotend (on the left) can't reach (right side)

**Expected results**
The tower should be on a pritnable area

mahtDFR, 2020-01-20:
As a workaround, you can edit the position of the prime tower by inputting custom X and Y positions.
Please upload a zipped 3mf of your project and log files too, as stated in the issue template. Then I can try to reproduce your issue.

Liger0, 2020-01-20:
The circular arrow meant that I was using the preset automatically calculated default which is wrong instead of a fixed custom value I did subtract by 20 in the profile.
[purge tower outside.zip](https://github.com/Ultimaker/Cura/files/4086733/purge.tower.outside.zip)

mahtDFR, 2020-01-20:
Ah I see. I've reproduced the error successfully, so will investigate further.

Ghostkeeper, 2020-01-21:
Fixed it. It was indeed not taking the nozzle offsets into account for the default value in fdmprinter. We didn't notice this because we override the prime tower position for our printers anyway, due to the bed clips that Ultimaker printers have in that corner.

closed 2020-01-21 by Ghostkeeper

Liger0, 2020-01-21:
Just to be sure since I have no idea how the software works, but I did read in the change done the line
`max(map(abs, extruderValues('machine_nozzle_offset_x'))`
and I am unsure what the abs function/variable does, so as I did put a project that was using an ABS material I got to think about it.
This fix will work on all materials, right?
Thank you.

fieldOfView, 2020-01-21:
`abs()` in most programming languages is a function to calculate the "absolute value" of what you pass to it; abs(-10) = 10, abs(4) = 4. In other words, it remove the minus sign from the value if there is one.

To set your mind at ease: it has nothing to do with the material named Acrylonitrile Butadiene Styrene here.

Ghostkeeper, 2020-01-24:
I documented the steps in that function in the commit message, since it's not immediately apparent even to a programmer. Basically this `abs()` function is called on all of the extruder values for the X offset, so they are all positive then. And then it takes the highest of them.

commit 5d21872e (2020-01-21): Take nozzle offsets into account when placing prime tower

extruderValues('machine_nozzle_offset_x') := [0, 20, -18]
map(abs(extruderValues('machine_nozzle_offset_x') := [0, 20, 18]
max(map(abs(extruderValues('machine_nozzle_offset_x') := 20

So we take the highest offset of all extruders to get the area that can be reached by all extruders. And we take the abs() of all extruder values because positive or negative only means that the other extruders get offset in the same direction.

Fixes #6997.