Ultimaker/Cura #6998: Circle geometry damaged in slicing process

crysxd, 2020-01-18:
**Application version**
4.4.0

**Platform**
macOS 10.15.2

**Printer**
Ender 3  Pro

**Reproduction steps**
1. Open attached .3mf file
2. Click "Open" in "Open Project" dialog
3. Click "Slice"
4. Select "Preview"

**Screenshot(s)**
[image]
[image]

**Actual results**
3 holes are properly sliced keeping the circle geometry but one hole is damaged

**Expected results**
All 4 holes should be round

**Project file**
[Cura_Bug.3mf.zip](https://github.com/Ultimaker/Cura/files/4080113/Cura_Bug.3mf.zip)

smartavionics, 2020-01-18:
The problem is the model is not valid. Install the Cura Mesh Tools from the Marketplace and then you can check yourself.

[image]

crysxd, 2020-01-18:
Fair point! Maybe a simple check like this would be a good feature for a future version of Cura. 

Thanks for the quick feedback!

closed 2020-01-18 by crysxd