Ultimaker/Cura #6999: 4.4.1 Crash on Start-up

ajmilner85, 2020-01-18:
**Application version**
4.41.

**Platform**
Windows 10 1809

**Printer**
Anycubic Mega S

**Reproduction steps**
1. Install Cura 4.4.1 - First use after clean install works fine
2. Restart system
3. Run Cura - Crashes on startup

**Project file**
N/A

**Log file**
2020-01-18 03:55:11,963 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin ConsoleLogger
2020-01-18 03:55:11,977 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin CuraDrive
2020-01-18 03:55:11,996 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [78]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.4\CuraEngine.exe
2020-01-18 03:55:11,998 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin CuraEngineBackend
2020-01-18 03:55:12,007 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin CuraProfileReader
2020-01-18 03:55:12,015 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin CuraProfileWriter
2020-01-18 03:55:12,024 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin OBJReader
2020-01-18 03:55:12,033 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin OBJWriter
2020-01-18 03:55:12,044 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin STLReader
2020-01-18 03:55:12,054 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin STLWriter
2020-01-18 03:55:12,062 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin FileLogger
2020-01-18 03:55:12,074 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin FirmwareUpdateChecker
2020-01-18 03:55:12,083 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin FirmwareUpdater
2020-01-18 03:55:12,093 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin GCodeGzReader
2020-01-18 03:55:12,102 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin GCodeGzWriter
2020-01-18 03:55:12,111 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin GCodeProfileReader
2020-01-18 03:55:12,126 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin GCodeReader
2020-01-18 03:55:12,137 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin GCodeWriter
2020-01-18 03:55:12,149 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin ImageReader
2020-01-18 03:55:12,159 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin LegacyProfileReader
2020-01-18 03:55:12,169 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin LocalContainerProvider
2020-01-18 03:55:12,180 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin LocalFileOutputDevice
2020-01-18 03:55:12,190 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin MachineSettingsAction
2020-01-18 03:55:12,200 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin ModelChecker
2020-01-18 03:55:12,210 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin MonitorStage
2020-01-18 03:55:12,222 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin PerObjectSettingsTool
2020-01-18 03:55:12,232 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin PostProcessingPlugin
2020-01-18 03:55:12,242 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin PrepareStage
2020-01-18 03:55:12,252 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin PreviewStage
2020-01-18 03:55:12,265 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin RemovableDriveOutputDevice
2020-01-18 03:55:12,283 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin SimulationView
2020-01-18 03:55:12,295 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin SliceInfoPlugin
2020-01-18 03:55:12,305 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin SolidView
2020-01-18 03:55:12,317 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin SupportEraser
2020-01-18 03:55:12,335 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [116]: No auth data to retrieve the access_token from
2020-01-18 03:55:12,337 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin Toolbox
2020-01-18 03:55:12,349 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin CameraTool
2020-01-18 03:55:12,362 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin MirrorTool
2020-01-18 03:55:12,376 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin RotateTool
2020-01-18 03:55:12,390 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin ScaleTool
2020-01-18 03:55:12,400 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin SelectionTool
2020-01-18 03:55:12,414 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin TranslateTool
2020-01-18 03:55:12,425 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin TrimeshReader
2020-01-18 03:55:12,439 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin UFPReader
2020-01-18 03:55:12,451 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin UFPWriter
2020-01-18 03:55:12,463 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin UltimakerMachineActions
2020-01-18 03:55:12,548 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin UM3NetworkPrinting
2020-01-18 03:55:12,566 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin UpdateChecker
2020-01-18 03:55:12,566 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [39]: Checking for new version of cura
2020-01-18 03:55:12,729 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin USBPrinting
2020-01-18 03:55:12,809 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin VersionUpgrade21to22
2020-01-18 03:55:12,837 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin VersionUpgrade22to24
[... 115 lines cut ...]
2020-01-18 03:55:18,944 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [34]: Updating CustomQualityProfilesDropDownMenuModel.
2020-01-18 03:55:18,952 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [63]: Updating QualityProfilesDropDownMenuModel.
2020-01-18 03:55:19,403 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [63]: Updating QualityProfilesDropDownMenuModel.
2020-01-18 03:55:19,409 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [34]: Updating CustomQualityProfilesDropDownMenuModel.
2020-01-18 03:55:19,757 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [181]: Support for Vertex Array Objects: True
2020-01-18 03:55:19,759 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [111]: Initialized OpenGL subsystems.
2020-01-18 03:55:19,760 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: OpenGL Version:  4.1.0 NVIDIA 441.66
2020-01-18 03:55:19,762 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Vendor:   NVIDIA Corporation
2020-01-18 03:55:19,763 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Renderer: GeForce RTX 2080 Ti/PCIe/SSE2
2020-01-18 03:55:19,765 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: GLSL Version:    4.0.0
2020-01-18 03:55:19,767 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.4\resources\shaders\default.shader]...
2020-01-18 03:55:19,769 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.4\resources\shaders\selection.shader]...
2020-01-18 03:55:19,771 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.4\resources\shaders\select_face.shader]...
2020-01-18 03:55:19,773 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.4\resources\shaders\default.shader]...
2020-01-18 03:55:19,775 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.4\resources\shaders\composite.shader]...
[... cut ...]
2020-01-18 03:55:19,788 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.4\resources\shaders\default.shader]...
2020-01-18 03:55:19,790 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.4\resources\shaders\grid.shader]...
2020-01-18 03:55:19,792 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.4\resources\shaders\platform.shader]...

**Additional information**
I understand there are a couple of start-up crashes you are currently chasing. Hope this helps narrow down the cause.

ajmilner85, 2020-01-18:
Have also just discovered that deleting all files in the user/AppData/local/cura and user/AppData/Roacming/ folders seems to resolve the issue. Cura runs fine on next attempt.

ajmilner85, 2020-01-18:
So, after deleting the relevant AppData folders I am able to close and run Cura successfully any number of times. 

However, as soon as I close Cura while the _window is open on my secondary monitor_, subsequent attempts to run the program fail. If I delete the user data the application runs normally again on my primary monitor.

Seems there is some issue rendering the application on the secondary monitor in some cases.

fieldOfView, 2020-01-18:
In the general preferences, there’s an option to “Restore the window position on start up” (or similar). Unselect that, and Cura should always open on your primary screen.

mahtDFR, 2020-01-20:
In response to the original report, this is a known issue that we are working on and hope to have it resolved in the next release.

Ghostkeeper, 2020-01-21:
> However, as soon as I close Cura while the window is open on my secondary monitor, subsequent attempts to run the program fail. If I delete the user data the application runs normally again on my primary monitor.

This indicates to me that this is a different issue than the others and fieldOfView's solution is probably a fix for him. Did you try that, ajmilner85?

Mikowmer, 2020-01-28:
While I'm not @ajmilner85, I do have this exact same issue, with an almost identical log file. I am running 2 monitors, and I can confirm that @fieldOfView's solution did work. If it attempts to open on the secondary monitor, Cura will freeze after loading the platform.shaders file.

I was wondering if there may be a glitch when it comes to interactions with graphics drivers on laptops, because many laptops with a dedicated graphics cards will also have integrated graphics on the CPU, and one will run the main display while the other will run secondary displays. Having done so, I believe I may have found useful information on the issue.

------

For information, I am running an NVIDIA GEFORCE RTX2060 with an 8th Gen Intel Core i7. My laptop is the MSI GL63 8SE if you want more specific stats. All drivers are up to date. Windows is as up to date as I can get it.

By default, Cura launches running on the integrated Intel graphics. When it opens with NVIDIA's graphics, while it does run, something seems odd: A frame counter appears in the corner (NVIDIA's overlay, no issue there, but when I open File, Edit, View, etc. the frame counter appears there _as well_. This will become important in a moment.

When it attempts to open on the secondary screen, whether running on Intel or RTX graphics, it will crash, as above. When I open it running Intel graphics on the Primary monitor, and then move it over to the Secondary monitor, it will work perfectly fine... until I try to open File, Edit, View, etc. at which point it crashes. Monitoring of the log files during this shows no additional logs as of the moment of the crash.

It appears that Cura freezes when opening a window on the secondary monitor for some unknown reason. I cannot isolate it to NVIDIA's drivers, but it is a possibility and maybe worth checking to see if the problem can be replicated on an AMD system.

cryptonobo, 2020-01-29:
Guys since 4.3 impossible to load Cura same with 4.4.1 here's the 4.3 THE ONE I CAN SEE LOG

Traceback (most recent call last):
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/IntentCategoryModel.py", line 101, in _update
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/IntentModel.py", line 40, in __init__
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/IntentModel.py", line 66, in _update
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/IntentModel.py", line 105, in _getActiveMaterials
KeyError: 'empty_material'

Thread 0x000070001039b000 (most recent call first):
  File "/Applications/Ultimaker Cura 2.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 78 in _updateThread
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000fe15000 (most recent call first):
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000f912000 (most recent call first):
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 549 in wait
  File "/Applications/Ultimaker Cura 2.app/Contents/MacOS/plugins/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000f40f000 (most recent call first):
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000ef0c000 (most recent call first):
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000ea09000 (most recent call first):
  File "/Applications/Ultimaker Cura 2.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000e506000 (most recent call first):
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 150 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000e003000 (most recent call first):
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 138 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000d977000 (most recent call first):
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 112 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000d474000 (most recent call first):
[... cut ...]
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

Ghostkeeper, 2020-01-30:
@cryptonobo this has been fixed in version 4.5. We found a couple of these crashes reported via the crash reporting screen (the "send report" button there).

Since the issue is worked around by enabling that option to not remember the window position, I'll close this issue as a duplicate of https://github.com/Ultimaker/Cura/issues/5651.

closed 2020-01-30 by Ghostkeeper