Ultimaker/Cura #7000: [DOCS] Software docs unable to resolve links to other support pages when pihole is blocking Google Tag Manager

mctoddenstein, 2020-01-18:
**Application version**
Website

**Platform**
Windows 10. Browsers Firefox and Chrome. Seems browser agnostic for the issues

**Reproduction steps**
1. Tried using search engine for "Infill Settings for Cura" and it dropped me at the Cura 3 "Infill settings" page via Firefox and DuckDuckGo (DDG). 
2. Attempted to click on the Cura 4 settings and it would not open the dropdown for the tree below Cura 4.
3. Attempted to click on any other setting in Cura 3's tree and it would also not open that page.
4. Attempted to access through Chrome and Google instead of Firefox and DDG to see if the issue was with either of those. Issue persisted and was able to be repro'ed in different browser. 
5. Testing the same page on my phone on LTE rather than local network shows that this works.

**Actual results**
Unable to access Cura's docs at my computer, where all work for slicing and printing is done

**Expected results**
Docs page should be able to be accessed without worrying about security and tracking from integration with google tag manager. More and more people are trying to secure themselves however possible. Having a pihole set up to block ads and tracking cookies should not break the ability to use a site.

**Log file**
HAR dump from the page showing that Google Tag Manager is not functioning correctly
[ultimaker.com_Archive [20-01-18 15-53-09].har.zip](https://github.com/Ultimaker/Cura/files/4081765/ultimaker.com_Archive.20-01-18.15-53-09.har.zip)

Console export showing Google Tag Manager script not loading for the page.
[console-export-2020-1-18_16-7-29.txt](https://github.com/Ultimaker/Cura/files/4081767/console-export-2020-1-18_16-7-29.txt)

mahtDFR, 2020-01-20:
Thanks for the report, but this is not the place to report issues for Ultimaker.com, this is the place to report issues for Cura.
I happen to know the answer to this anyway so I will say that this feature is not going to be worked on since we will switch over to a new system at some point very soon, which will have even better functionality than a fix for this issue.

closed 2020-01-20 by mahtDFR

mctoddenstein, 2020-01-20:
@mahtDFR Thank you for the response. I didn't know if this was the place to report for Ultimaker.com and I appreciate your clarification on that. Thank you for the info that Ultimaker will be switching to a new system soon as well. I really enjoy the software and getting clarification on how a function works, or is supposed to work, is very helpful for me. Not having that easily available because of the underlying system and users' desire for privacy was a little upsetting, but I understand that it is how Ultimaker created the website. Not a lot I can do on that front.

Sorry if I wasted anyone's time, but I do appreciate the response on this. I'll keep an eye out for any funny business in Cura and report it in the future!

Ghostkeeper, 2020-01-21:
I'm also blocking Google Tag Manager (using uMatrix in Firefox) but the website works fine for me.