Ultimaker/Cura #7001: Candle 1.2 After loading gcode nothing showing in vizualiser

cwtears, 2020-01-20:
**Application version** 1.2
(The version of the application this issue occurs with.)

**Platform** Win 10 Pro
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1.Loaded a Gcode file
2. Everything loads but nothing displays in the vizualizer window.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)
I am new to CNC and as I am also in my 70's please excuse me if I have made a Nebie mistake.

fieldOfView, 2020-01-20:
Unless you mistyped "Cura", I think you are in the wrong issue queue here. Candle is a different application: https://github.com/Denvi/Candle

mahtDFR, 2020-01-20:
Wrong software, sorry!
As mentioned above it looks like you need to report an issue in the Candle repo https://github.com/Denvi/Candle
I will close this ticket as it's not relevant to us.

closed 2020-01-20 by mahtDFR