Ultimaker/Cura #7004: [Running from Source] Crash on STL loading

alexwoo1900, 2020-01-20:
**Cura**
4.4 branch

**Uranium**
4.4 branch

**Python**
3.7.1

**PyQt5**
5.1.0

**Platform**
Windows 7 64bits

**Actual results**
1MB Ironman.stl (ok)
7MB Ace.stl (sometimes crash)
56MB Nami.stl (always crash)

**Log**
2020-01-20 18:50:54,623 - DEBUG - [Thread-2] UM.Mesh.MeshData.calculateNormalsFr
omVertices [482]: Calculating normals took 0.13500761985778809 seconds
2020-01-20 18:50:54,683 - DEBUG - [Thread-2] STLReader.STLReader._read [97]: Loa
ded a mesh with 3541878 vertices
2020-01-20 18:50:55,539 - DEBUG - [Thread-2] UM.Mesh.MeshData.approximateConvexH
ull [439]: approximateConvexHull(target_count=1024) Calculating 3D convex hull t
ook 0.8540487289428711 seconds. 3717 input vertices. 3717 output vertices.
Windows fatal exception: access violation

Windows fatal exception: Thread 0x000020b0 (most recent call first):
  File access violation

Windows fatal exception: access violation

"D:Windows fatal exception: access violation\alex_projects\Cura-4.4\cura\..\plug
i

ns\USBPrinting\USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 865 in run
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 917 in _bootstrap_inner
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 885 in _bootstrap

Thread 0x00001d8c (most recent call first):
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 300 in wait
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\site-packages\z
eroconf\__init__.py", line 2165 in wait
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\site-packages\z
eroconf\__init__.py", line 1466 in run
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 917 in _bootstrap_inner
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 885 in _bootstrap

Thread 0x00001f90 (most recent call first):
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 300 in wait
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 552 in wait
  File "D:\alex_projects\Cura-4.4\cura\..\plugins\UM3NetworkPrinting\src\Network
\ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 865 in run
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 917 in _bootstrap_inner
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
[... 126 lines cut ...]
Thread 0x00001a6c (most recent call first):
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 296 in wait
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 427 in acquire
  File "D:\alex_projects\Uranium-4.4\UM\JobQueue.py", line 86 in _nextJob
  File "D:\alex_projects\Uranium-4.4\UM\JobQueue.py", line 112 in run
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 917 in _bootstrap_inner
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\threading.py",
line 885 in _bootstrap

Thread 0x00001a64 (most recent call first):
  File "C:\Users\Alex\AppData\Local\Programs\Python\Python37\lib\site-packages\n
umpy\linalg\linalg.py", line 532 in inv
  File "D:\alex_projects\Uranium-4.4\UM\Math\Matrix.py", line 134 in invert
  File "D:\alex_projects\Uranium-4.4\UM\View\RenderBatch.py", line 202 in render

  File "D:\alex_projects\Uranium-4.4\UM\View\DefaultPass.py", line 23 in render
  File "D:\alex_projects\Uranium-4.4\UM\Qt\QtRenderer.py", line 147 in render
  File "D:\alex_projects\Uranium-4.4\UM\Qt\Bindings\MainWindow.py", line 232 in
_render
  File "D:\alex_projects\Uranium-4.4\UM\Qt\QtApplication.py", line 341 in exec_
  File "D:\alex_projects\Cura-4.4\cura\CuraApplication.py", line 804 in run
  File "cura_app.py", line 180 in <module>

nallath, 2020-01-20:
What version of numpy  (and numpy-stl?) are you using?

alexwoo1900, 2020-01-21:
@nallath 1.15.3

mahtDFR, 2020-01-21:
Our build uses numpy 1.15.4. Can you try running `pip install numpy==1.15.4`
and also `pip install numpy-stl==2.10.1` and report back?

mahtDFR, 2020-02-18:
Closing this off, assumed fixed

alexwoo1900, 2020-02-20:
@mahtDFR 
Sry for my late feedback.
Upgraded to 1.15.4 and 2.10.1 but no help. Same error information.
All is well in installation version.

Ghostkeeper, 2020-02-25:
Then I'm wondering what is different. I've installed a development environment on a bunch of computers but didn't have this issue yet.

Is it only happening with a certain model, maybe? Could you try ASCII-STL vs. binary-STL?

alexwoo1900, 2020-03-10:
@Ghostkeeper 
After testing, the key to all problem is model size. I cannot load a model larger than 10m. But small model is ok. 

Obviously, it's not about my Windows. My installation Cura works well. I think all about the software environment.

I found some people report on FB that they are also experiencing software crash when loading a STL.

alexwoo1900, 2020-03-10:
Full Log [Here](https://pastebin.com/n2r91XzT)

Ghostkeeper, 2020-03-13:
Is there an update available for Numpy or Scipy? It's crashing somewhere in a native (C/C++) library, or Python would raise an exception in your log. It's most likely one of those dependencies.

alexwoo1900, 2020-04-02:
@Ghostkeeper 

Numpy and Scipy are the latest(1.18.2 & 1.4.1) and the program still crashes. I saw it on other computers.

[Snapshot](https://drive.google.com/file/d/1uxP6uAVaO9MX93XU9GlLr5PyDyf1jyA6/view?usp=sharing) (no engine, just frontend)

Ghostkeeper, 2020-04-06:
Hrm, my browser keeps saying that there's a redirect loop in that download :(

I have the same versions of Numpy and Scipy, but I'm not able to reproduce the problem :( Maybe GDB would shed some light on this, but it's quite a hassle to set that up to break on the correct spot.

alexwoo1900, 2020-04-07:
**SOLVED**
For 64 bit systems, 64 bit numpy should be installed. **PIP installs 32-bit numpy by default!**

**HOW DID I FIND OUT**
I located the problem to 
"data = data.dot(transformation.getTransposed().getData())"
in function "transformVertices"

Access violation means that memory allcation of a variable changes before and after dot product.
This is most likely due to numpy's inability to properly handle 64 bit data.

**SOLUTION**
download and install 64 bit numpy from: 
https://www.lfd.uci.edu/~gohlke/pythonlibs/#numpy

**SUGGESTION**
Cura team should reminds developers to install numpy just like they install sharply library, rather than just using "python -m pip install numpy" like your wiki said.

mahtDFR, 2020-04-07:
Ah, thanks for sharing a solution! We can update the documentation to include this.

Ghostkeeper, 2020-04-08:
I've adjusted the installation instructions for Windows from source, on the wiki.

closed 2020-04-08 by Ghostkeeper