Ultimaker/Cura #7007: Show active support blocker areas by swapping overhang color.

Nandox7, 2020-01-20:
**Is your feature request related to a problem? Please describe.**
Not exactly a problem but a nuisance. 
Current support blocker adds "blocks" to the part. In smaller parts or parts that have only a few supports to block it works well but for bigger parts or ones with many supports that need to be blocked (e.g.: round holes) it gets quite messy to navigate the part as well as overall cura performance suffers a lot with it.

**Describe the solution you'd like**
Give option to show the support blocker in a more simple visual way. Changing the color of the overhang areas in the part.
Currently Cura (Default color palette) show this overhang areas where support will be added as a bright red. The feature I propose here would be that upon clicking on them, in the same way we now select with support blocker feature, they would change to a different color that would signal to the user that support is blocked there. 
Visually is more less messy as does not create as much clutter and it should be less demanding in terms of rendering/processing to software as well.

**Describe alternatives you've considered**
Current alternative is the one resent at the moment and that has eh limitations described above.

**Affected users and/or printers**
Target audience is all Cura users.

**Additional context**
Screenshot added to better visualize the idea.
Bright red: normal overhang areas detected by Cura.
Bright green: areas where the user clicked with the support blocker feature and for which support will not be generated.

mahtDFR, 2020-01-21:
We discussed briefly and we like the idea, but perhaps the colour selection should be altered a little. The support blocker should not disappear, otherwise how would you find it? Also, what if you have green filament? Or red filament for that matter - already a problem for us. We will discuss internally and see what it comes to.

Nandox7, 2020-01-21:
Thanks for considering this FR.

Yes the color can be an issue and something I realized too when thinking about this.
One alternative could be using instead a pattern, similar to what happens when Cura signals an object outside the print space.
Still possible to have some conflict with multiple color parts but far less likely and in terms of rendering (I assume) more heavy in terms of rendering due to the mapping.

nallath, 2020-01-24:
We've added this FR to our backlog (CURA-7138)