Ultimaker/Cura #7008: Multiple time/cost estimation info boxes; saved per slice

airsail-p9, 2020-01-20:
**Is your feature request related to a problem? Please describe.**
I always try to optimise a print  for the current requirements (speed, detail, robustness etc.) For this I usually do multiple slices to see the effect on print time and cost (especially when slicing for efficient prints). To see the effect of a change, I tend to do go back and forth to see the improvement, which is time-consuming. Sure, I could take a pen and write it down, but wouldn't a programmed solution be cooler? =)

**Describe the solution you'd like**
A solution would be to have the info-box of the last slices saved and displayed. The boxes could be iteratively named "slice 1", "slice 2" and so on and contain the info it currently contains plus the changes that were made in relation to the previous slice. 

**Describe alternatives you've considered**
pen and paper ;)

**Affected users and/or printers**
I think this can be useful for everybody!

**Additional context**
I attached an example how I image it to look like approximately (example quickly made with paint..)

[image]
[image]

It's my first idea for a feature, hope you like it :)

mahtDFR, 2020-01-21:
We discussed briefly but besides the code time it would take, we feel it needs a cleaner UX solution, the proposed one seems a little confusing to the user without explanation. If there is a clearer proposal, we can perhaps start work on it.

What if there were arrow buttons to cycle through previous slices, or click on the 'i' button to bring up the previous slices?

airsail-p9, 2020-01-22:
Glad you like it =)

Sure, I would appreciate a cleaner solution. I like the arrow solution! 

Another idea would be to have only the previous slice in a second window next to the other one (with a headline "previous slice") that is permanently displayed and updates when a new slice is done. I realised that it is probably enough to compare with the slice before, and this would be easier to code.

fieldOfView, 2020-01-22:
How about a floating panel that shows the undo states, like the History panel in Photoshop.

mahtDFR, 2020-01-24:
Good suggestion, it can be built, although it is a lot of development work. I will put to the team and see what comes back.

mahtDFR, 2020-01-27:
Thanks for the suggestions, but unfortunately we've decided to defer this as we cannot devote the development time to it right now.

airsail-p9, 2023-06-30:
Hey, I still think it would be a useful feature :) Would be happy if it could go back to be considered. Best regards!

airsail-p9, 2023-06-30:
Hi, I still think this would be an interesting feature! Best regards!

airsail-p9, 2023-06-30:
Pls reopen

airsail-p9, 2023-06-30:
Hi, I tried to reply in the forum but it's closed now.. I still think it would be an interesting feature.Cheers

fieldOfView, 2023-06-30:
It seems you inadvertently closed the issue yourself.
Let me reopen it for you.