Ultimaker/Cura #7009: Tabs aka multiple concurrent sessions

kr15uk, 2020-01-20:
Quite often I do work on multiple machines and/or "plan-ahead" projects at the same time therefore require to have multiple sessions so to speak.

My current workaround is terminal launch additional Cura instance as different user, which is not ideal.

Yes I know there is save project option but that's not the same as concurrent windows to flip between.

Alternatively opening additional Tab for next project/machine, or even full window would be nice for sure.

Yep, this might be quite niche request but whenever you have time if any.

Ta.

Kris

mahtDFR, 2020-01-21:
Interesting idea, probably quite complex to implement though. We will discuss.

mahtDFR, 2020-01-24:
Thanks for the feature request. Unfortunately we've decided not to follow up on it this time due to the development time it will take.

There is a semi-workaround in place if you have multiple machines connected by switching the machine tab, although I understand this functionality doesn't really answer your request completely.

closed 2020-01-24 by mahtDFR