Ultimaker/Cura #7014: M486 integration

maukcc, 2020-01-22:
Is there any talk about this?
https://reprap.org/wiki/G-code#M486:_Cancel_Object

mahtDFR, 2020-01-22:
It's something we've considered before. It seems it would work best over a serial connection? What use case did you have in mind for this?

maukcc, 2020-01-22:
when printing 20 objects at the same time, and object 8 is not going well, you would be able to discontinue object 8, while the others will finish .

mahtDFR, 2020-01-23:
Update on this: its up on the board for discussion as it seems to have a bit of interest. Will update when I know more as we need to work out a UX implementation too.

mahtDFR, 2020-01-24:
Thanks for the request, we've decided to pick this up. It will also require some firmware implementation so this will also need to be explored in parallel.

nallath, 2020-01-24:
I've added this to our backlog (CURA-7139)

Sophist-UK, 2020-05-23:
This feature request is to add Cancel Object to the USB printing functionality, See also #7812 which is to add support for the Octoprint Cancel Object function.

truglodite, 2021-07-04:
I was just looking at why the cancel objects menu on my printer is never populated during a multi part print... and realized Cura simply has not implemented M486 capabilities yet. It seems Prusa slicer already has a plugin that works for this. Would be nice if cura added this too.

Ghostkeeper, 2021-07-12:
The main reason why Cura didn't implement this is because Ultimaker's printers don't support the M486 command. So whatever we implement can't be tested at work.

shinmai, 2022-01-26:
The Python likely isn't up to Cura release standards and I don't have the resources to invest in to a proper PR process to take it there, but if anyone want's to use M486 with Cura right now, I've made a [very simple 15-line Post Processing script](https://github.com/shinmai/cura-M486) that'll inject the proper M486 G-codes to make things work with Marlin.

This assigns indices as meshes are encountered in the G-code, which might not always be very intuitive without referencing the G-code or it's preview, so I recommend enabling `CANCEL_OBJECTS_REPORTING` in Marlin to get LCD status messages that show the ID of the current object being printed to more easily identify objects.

closed 2023-07-13 by github-actions[bot]