Ultimaker/Cura #7016: Cura crashes my entire computer

CLUBJACOB1, 2020-01-23:
**Application version**
4.41

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Ryzen 7 2700X

RTX 2080

32GB RAM  WINDOWS 10 
**Printer**
ENDER PRO 3 

**Reproduction steps**
INSTALLED OLDER VERSION AND BETA VERSION 

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
PC LOCKS UP BLACK SCREEN AND POWER AND REST BUTTONS DO NOT WORK 

**Expected results**
I REINSTALLED AND RANDOMLY RESTART OR BLACK SCREEN 

**Project file**
ANY FILE JUST RANDOMLY TURNS OFF 
**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I pulled my OC off the CPU and GPU still am getting blackscreened after about 30 minutes in CURA.

BIOS are updated and I have checked CPU voltage and made sure there isn't a power problem.
 Replaced every single part in my pc thinking it was defective shit  Rig would crash anywhere from minutes to hours after closing cura. I just think AMD bugs out with it for some reason, Black screen, unresponsive had to hard reset from PSU, but sometimes it would randomly reboot itself. Still happens with 4.1 btw

mahtDFR, 2020-01-23:
Thanks for the report, It's hard to diagnose an issue with the information you have given, can you please send a log?
https://github.com/Ultimaker/Cura#logging-issues

And if possible a 3MF file or project file you have used when a crash occured?

fieldOfView, 2020-01-23:
Likely a duplicate of https://github.com/Ultimaker/Cura/issues/5265

closed 2020-01-24 by mahtDFR