Ultimaker/Cura #7017: [master] mouse view movement very jerky when settings visible

smartavionics, 2020-01-23:
**Application version**
master

**Platform**
Linux (Qt 5.13.2)

**Printer**
Any

**Reproduction steps**
Load some model onto the build plate, make the settings visible and then move/rotate it using the mouse. The view moves quite jerkily. Now close the settings box and the view can be moved/rotated much more smoothly.

**Screenshot(s)**

Here's an example that will not move/rotate smoothly with the mouse...

[image]

**Actual results**
Model view moves/rotates jerkily.

**Expected results**
Model view should be smooth.

**Project file**
Any reasonably complex file.

**Log file**

**Additional information**

I noticed this when testing the RawMouse plugin using the 3d mouse. When the 3d mouse is used, the view is moved/rotated/zoomed completely smoothly while using the mouse is very jerky.
I thought it was some bad side effect of the RawMouse plugin but removing that makes no difference. What does make a difference is closing the settings box. With that hidden, the view can be manipulated with the mouse more smoothly.

It's almost as if a timer is going off every second or two and causing some load or latency?

ninovanhooff, 2020-01-23:
@evtrados Can you please try on an X1 laptop? I tried 
thingiverse.com/thing:22051
on a powerful system and did not see slowdowns with settings visible vs invisible

nallath, 2020-01-23:
Well, it does make sense that there is a difference. If you rotate the camera, you do a full re-draw of everything on the screen (including the visible QML items). So if you decrease the amount of items on the screen, the re-draw should be faster.

smartavionics, 2020-01-23:
Yes, but when I use the 3d mouse to rotate/zoom/pan the view the movement is completely smooth irrespective of whether the settings box is visible or not. In terms of the scene redraw, they are doing the same actions.

nallath, 2020-01-23:
Yeah, there is something going on there. I think the redraw events for the camera are being fired twice (since the issue does not happen for me if zoom with the camera...)

smartavionics, 2020-01-23:
Could be a good incentive to buy a 3d mouse !

smartavionics, 2020-01-24:
I am now thinking that the settings being visible or not is actually irrelevant and that the problem is simply that when using the (2d) mouse to move or rotate the view the motion is not always smooth. As mentioned above, it may just be worse when the settings are visible as there is more to redraw.

nallath, 2020-01-24:
I think I've found what was causing it. Could you try the latest master again?

smartavionics, 2020-01-24:
Sorry, I don't think that helped. The problem is most noticeable when moving the view using the middle mouse button. As I move the mouse, from time to time the view "sticks" and then jumps across to where it should be. Rotation (using the right button) doesn't seem to be affected anywhere nearly as badly.

nallath, 2020-01-24:
Allrighty, back to my investigation!

smartavionics, 2020-01-24:
It gets more interesting...

Without the yappi profiler running, moving the view with the middle mouse button is not smooth. When I run the profiler, the movement slows down a bit because of all of the profiling going on but it is smooth and doesn't jump about. As soon as I stop the profiler, it becomes "lumpy" again.

Anyway, here's the profile file. All I did was move the view and also a bit of rotation.

[curaProfile.zip](https://github.com/Ultimaker/Cura/files/4108927/curaProfile.zip)

nallath, 2020-01-24:
I don't know what the hell is causing this. It being a heisenberg issue is also not making it any easier to debug it.

smartavionics, 2020-01-24:
Are you actually seeing the jumpy movement there or is it just me?

I wonder if it is due to the _moveCamera() routine scaling the move distance with the event delta values and so if (for some reason) the UI doesn't process the mouse movements at a uniform rate, the deltas will vary a lot in magnitude which would make the movement not smooth ?

nallath, 2020-01-24:
I'm not seeing it. I assumed it to be a timing / performance issue, so that was what I was trying to figure out. The reason that the delta between them isn't smooth is usually because the time between the frames isn't the same. 

Anyway, the _moveCamera uses the same delta stuff as the _rotateCamera, and the rotate doesn't cause (as much) issues for you, yet the _moveCamera is.

smartavionics, 2020-01-24:
Well, if you can't see it happening it's not worth you worrying about it for now. Let's keep this issue open for a bit and if I can provide any more info, i will report back.

smartavionics, 2020-07-14:
Current master branch doesn't appear to have this issue any more. Closing...

closed 2020-07-14 by smartavionics

Ghostkeeper, 2020-07-15:
This may have been fixed by the rendering improvements we've made. Since 4.6 it should no longer re-render the interface elements when the 3D scene is re-rendered.