Ultimaker/Cura #7018: [4.4.1] Support partially not generated

phosphide, 2020-01-23:
**Application version**
4.4.1

**Platform**
macOS 10.14.6

**Printer**
Ultimaker 3

**Reproduction steps**
1. Load attached STL
2. Enable PVA support with second extruder
3. Slice, look at preview

**Screenshot(s)**
[image]

**Actual results**
Circular cutouts at the bottom of the part are partially not supported (see bottom left of screenshot).

**Expected results**
Whole circles should be supported.

**Project file**
[iot-hvac bottom.stl.zip](https://github.com/Ultimaker/Cura/files/4105126/iot-hvac.bottom.stl.zip)

**Additional information**
The cutouts are 0.5 mm high, I tested with both built-in PLA+PVA/fine 0.1 mm profiles and my custom ones.
I experimented with support pattern, distances, expansion, overhang angle, etc. A change in the settings almost always changes the shape of missing support sections, but the problem persists.

I had a similar problem with other parts before, but this one is especially annoying. The part is designed in Fusion 360, the cutouts are just simple 10.7 x 0.5 mm circular extrusions.

rburema, 2020-01-24:
This seems to be related to our stair stepping bug.

The workaround is to set 'Support Stair Step Maximum Width' to a low value, like 1.0 (and/)or set 'Support Stair Step Height' to 0.

Closing this as a duplicate of # please continue the discussion there.

closed 2020-01-24 by rburema

rburema, 2020-06-25:
We might finally have squashed this bug! When we eventually release 4.7 (which might take a bit longer than usual) you can check it out.