Ultimaker/Cura #7019: 4.4.1 layout bug

jellewie, 2020-01-24:
All of a sudden my CURA 4.2 it's layout was broken,

I unstilled all of cura, and installed thenew 4.4.1 version to be supriced with the same layout after an hour to put al my settings back in to it.
[https://i.imgur.com/d0R1qUr.png](https://i.imgur.com/d0R1qUr.png)

**Application version**
4.4.1

**Platform**
Win10 64bit GTX1050 mobile

**Printer**
Ender 3

**Reproduction steps**
1. install all
2. install new 4.4.1
3. install settings and plugins

**Screenshot(s)**
Win10 64bit

**Actual results**
layout got fucked

**Expected results**
that the layout doesnt look like crap

**Project file**
https://drive.google.com/open?id=1Te1wgdeuHPjJDNVjQs6vqG9D5LneUUrj

**Log file**
https://drive.google.com/open?id=1dGzUe9AugDvESaSxVqxoR6GG-9tiXtsR

mahtDFR, 2020-01-27:
Couldn't replicate the situation with the files project files you provided but your log gives some clues to the issue:

``2020-01-24 17:47:34,014 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: Traceback (most recent call last):
2020-01-24 17:47:34,016 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 230, in _render
2020-01-24 17:47:34,019 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtRenderer.py", line 118, in beginRendering
2020-01-24 17:47:34,022 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtRenderer.py", line 183, in _initialize
2020-01-24 17:47:34,024 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGL.py", line 46, in __init__
2020-01-24 17:47:34,029 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: RuntimeError: Try to create singleton 'OpenGL' more than once``

Qt is the tool the frontend is built in, and it's throwing up critical errors.

Something to try: in Windows Exporer, type ``%localappdata%/cache`` in the address bar and delete the ``qtshadercache`` folder, then restart Cura.

Hope this helps

jellewie, 2020-01-27:
That seemed to have fixed the issue,

I was looking through the logs and your comment and 'python' was mentioned a lot.
Lately, I had some problems with PlatformIO and python and I de-installed python completely and reinstalled just the latest version though PIO. Also some windows references to old python locations have changed. This exactly overlaps with the timeframe of my Cura problems. Could this have caused it?

Although the symptoms have been fixed, and everything seems to work again, I will leave the ticket open since the problem hasn't really be found/solved (Aka; Cura should throw an error, or identify and fix the problem by itzelf) But if an admin doesn't agree, it could be closed

Ghostkeeper, 2020-01-27:
> Could this have caused it?

Cura comes with its own Python installation, so installing or deinstalling other Python version shouldn't™ affect it. This problem in particular has more to do with Qt (the framework that draws the application) rather than Python (the code that orchestrates everything the front-end does).

The problem we're facing here is that this `qtshadercache` is not mentioned in any of our code. If it gets corrupt, it is because the Qt framework allows it to become corrupt, and there's not a lot we can really do. The only thing we can really do is to try and work around the bug in Qt so that the offending code doesn't get caught. This is fairly hard but can be achieved.

That folder itself is not something we should touch either, as it's not a folder that's specific to Cura but is shared with all applications that use the Qt framework as well, such as Google Earth, Valve's Source engine, 3DS Max or Photoshop Elements. If we meddle with the folder we could break those applications. Also, it could be that any of those programs corrupted the cache, not Cura. So it's quite a complex issue in that regard. We can't just clear the folder on every launch.

closed 2024-12-07 by github-actions[bot]