Ultimaker/Cura #7020: Cura 4.4.1 wont print brim with 2nd extruder.

TCPILOT, 2020-01-24:
**Application version**
Cura 4.4.1

**Platform**
Windows 10

**Printer**
Bibo 3D Dual

**Reproduction steps**
1. I sliced file for dual extrusion.
2. Set brim and supports to be printed with 2nd extruder.

**Actual results**
Printer only prints brim with extruder 1 no matter what I do.

**Expected results**
Printer should print the brim with extruder 2 and not extruder 1.

Ghostkeeper, 2020-01-27:
Can you provide a project file that reproduces your problem? You can save a project file using File -> Save...

If I change the Build Plate Adhesion Extruder setting to the 2nd extruder, it properly changes the brim to the other extruder. Though of course it will still prime the 1st extruder as well by printing a skirt around that brim.

mqbenedict, 2020-05-07:
I have the same problem using a Bibo dual extruder printer and Cura 4.6.0 on a MacBook. The brim is supposed to print on extruder 2 but it prints on 1. Attached is a file with this issue.

mahtDFR, 2020-05-07:
> Attached is a file with this issue.

Sorry, can't see an attachment here. Can you try again? You'll need to do it via GitHub rather than email for it to work.

mqbenedict, 2020-05-07:
Sure thing. BTW: I am printing the infill with extruder 2 and the shell with 1 and that works fine.

Ah, I see I can't upload the gcode file. I'll rename it 'txt'.
[BD_companionway brackets Right.txt](https://github.com/Ultimaker/Cura/files/4593763/BD_companionway.brackets.Right.txt)

mahtDFR, 2020-05-07:
Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so I can try to reproduce it with the same settings as you.

mqbenedict, 2020-05-07:
You got it.

Mark
[Project with Brim issue.3mf.zip](https://github.com/Ultimaker/Cura/files/4593794/Project.with.Brim.issue.3mf.zip)

mqbenedict, 2020-05-07:
Just noticed another problem with the program. Extruder 1 which is incorrectly being used to print the brim is at its standby temperature (175) rather than the set print temperature (200). Extruder 2 (which should be used) goes to printing temperature (200) rather than its standby temperature (175) but is not used.

mahtDFR, 2020-05-08:
When I open your project file I'm greeted with this:

[image]

Is this a pre shipped profile, or one you've added manually? If you've added it manually there's little we can do to help there.

> Extruder 1 which is incorrectly being used to print the brim...Extruder 2 (which should be used)

According to your project file, you have set the brim to print with E1 and not E2. Did you adjust this setting?

[image]

mqbenedict, 2020-05-08:
That's strange. Attached are two screenshots of the same project I sent you (BD_Project with Brim issue). You can see that when either extruder 1 or 2 is highlighted, it says 'extruder 2'. Very puzzled. Not sure how you could see one thing and I see something different.

Thanks for your help.

Mark B.

(Yes, I made a mistake on the alignment on the bed.)

[image]
[image]

mahtDFR, 2020-05-08:
It's possible that that setting got switched back because the machine profile changed - just wanted to make sure you were aware of that setting in the first instance.

Another thing you could try - can you share your [user data](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations) with me? Then I have the same conditions as you. (You probably need to use WeTransfer due to GitHub size limitations).

Ghostkeeper, 2020-05-11:
> When I open your project file I'm greeted with this:
> 
> [image]
> 
> Is this a pre-shipped machine profile, or one you've added manually? If you've added it manually there's little we can do to help there.

The Bibo 2 Dual is shipped with Cura. For me this project file loads fine!

Ghostkeeper, 2020-05-11:
According to Cura it prints the brim and support properly with the second extruder:

[image]

However the start g-code of the Bibo 2 changes that to the first extruder (the `;;;;` comment is mine):

```
G21        ;metric values
G90        ;absolute positioning
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G1 Z2.0 F400 ;move the platform down 15mm
T0       ;;;; <-- Hard-coded extruder number!
G92 E0
G28
G1 Y0 F1200 E0
G92 E0
M117 BIBO Printing...
```

I'd suggest changing that line from `T0` into `T{initial_extruder_nr}`.

closed 2020-06-19 by no-response[bot]