Ultimaker/Cura #7021: Cura [4.4.1] only runs once, on fresh restart of computer

thedanwray, 2020-01-24:
**Application version**
4.4.1

**Platform**
Windows 10, HP Laptop on dock with 2 external monitors, lid closed (laptop screen not used)

**Printer**
Any

**Reproduction steps**
1. Boot, or restart computer.
2. Run cura
3. Close cura
4. Try to run cura again, never starts, no splash screen, only appears in task manager.
5. Restart Computer.
6. See step 2

**Screenshot(s)**
Nothing to show, it either runs or it doesnt

**Actual results**
This is explained in the steps

**Expected results**
It should run
**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
Tried including, but was given error: There was an error creating your Issue: body is too long (maximum is 65536 characters).

**Additional information**
(Extra information relevant to the issue.)

Ritson1234, 2020-01-26:
having the same issue. fresh install and opens. close and reopen and fails to open. windows 10

Ghostkeeper, 2020-01-27:
We fixed this for Cura 4.5. There's also a workaround: Update your Intel video drivers. Intel made a fix for their driver bug like 2 weeks ago.

Technical details: There was a bug in the Intel video drivers that made Cura freeze when it deletes an OpenGL context (basically an environment that it can render 3D things in, knows where to send the 3D data, etc). We create and delete two OpenGL contexts at the start of the application in order to detect which OpenGL version is supported. In Cura 4.5, we're just not deleting that unused extra context any more which prevents the crash (though it takes slightly more memory).

closed 2020-01-27 by Ghostkeeper

fieldOfView, 2020-01-27:
>In Cura 4.5, we're just not deleting that unused extra context any more which prevents the crash (though it takes slightly more memory).

Almost sounds like the fix is worse than the problem. Can this fix of leaking resources by design somehow be limited to only the affected platform and driver? Because in a couple of months the driver will be mostly something of the past, and Cura would still be leaking resources.

Ghostkeeper, 2020-01-27:
@rburema was this fix in the end implemented as an option only or was it always enabled?

thedanwray, 2020-01-27:
So where do I get cura 4.5?

fieldOfView, 2020-01-27:
It does not yet exist.

rburema, 2020-01-28:
@Ghostkeeper @fieldOfView As a side-effect of the other attempted fixes (still deemed useful in the future), you _can_ prevent the 'leakage' (which only happens a single time ... still bad, but not as bad as hanging for a growing percentage of users IMHO) by manually writing `force_modern_opengl = True` under the `[view]` header to the preferences.

I suppose I could set the 'leaked' context to `None` (which'll garbage collect it) if the platform isn't windows and the drivers aren't intel?

Ghostkeeper, 2020-01-29:
To be clear, the operating system still cleans up the extra OpenGL context when Cura is closed down. I don't know how much memory an empty context takes, (depends on the driver I guess) but I don't think it's going to be a whole lot since there is no graphics data in it. Maybe it'll hold an empty buffer or two.

> I suppose I could set the 'leaked' context to `None` (which'll garbage collect it) if the platform isn't windows and the drivers aren't intel?

We can do that, at least for MacOS and Linux. Is it easy to detect the driver manufacturer?

AntoineTurmel, 2020-01-29:
I had this issue (running on Intel uhd graphics 620), updating the drivers solved the issue for me:
https://downloadcenter.intel.com/download/29335/C-ur-graphique-Intel-Windows-10-DCH