Ultimaker/Cura #7023: [4.4.1] Profile settings aren't persisting between Cura Windows

sunlanterns, 2020-01-25:
**Application version**
4.4.1

**Platform**
Windows 10

**Printer**
Wanhao PowerSpec Duplicator I3 Mini

**Reproduction steps**
1. Open Cura
2. Open another window of Cura
3. Make a new profile in the second window
4. Profile does not appear in the first Cura window

**Screenshot(s)**
[image]

**Actual results**
The new profile does not show up in the original Cura window

**Expected results**
The new profile should show up in the original Cura window

fieldOfView, 2020-01-25:
This is expected behavior. Do you know of any application where running two copies simultaneously does synchronize settings like this?

sunlanterns, 2020-01-27:
Microsoft word does this. It opens documents as completely separate windows but if you make a new text style in one, it immediately shows up in the other.

Ghostkeeper, 2020-01-27:
Firefox and Chrome are the only ones that I can think of (and now Word as well), but all three of these actually run single-instance in the background and just have multiple different windows for the same process that they all update. That in itself is something we can consider to prevent corruption issues, but I think that your feature request is too much work for us to implement for what you get out of it.

You can break this functionality for Firefox by running one of the instances as administrator by the way. It can't run it through the same process any more then and it'll break the syncing of your preferences across all windows.

sunlanterns, 2020-01-27:
@Ghostkeeper 
Would it be an easier ask to have an 'update settings' button such that when clicked, Cura would reload in the settings files without having to completely reload the Cura software?

Ghostkeeper, 2020-01-29:
Not really. To implement your original feature I'd watch for file system changes and add individual files one by one as if you're importing them, deleting the old one if they existed in memory. But to implement a reload button we'd need to temporarily disable the interface, reload all containers and rebuild the cache, and then rebuild all of the lists of available profiles. It's basically 70% of the start-up time of Cura, just loading all of the profiles and making them ready for display in the interface.

closed 2023-06-13 by github-actions[bot]