Ultimaker/Cura #7024: [4.4.1] Support Towers without normal supports

sunlanterns, 2020-01-25:
**Is your feature request related to a problem? Please describe.**
There are some models that have a piece that would require a support tower underneath of it, as well as other areas that don't touch the build plate that require no supports, such as this model from thingiverse (presuming you don't want to use the version with the support tower the original maker provided):
https://www.thingiverse.com/thing:3495390

Where the mouth could use a tower but the rest of the model doesn't need the supports that cura ends up generating.

**Describe the solution you'd like**
The ability to turn on towers without turning on supports in general.

**Describe alternatives you've considered**
I've tried playing with the support settings, like minimum area, but it tends to end up getting rid of the tower as well.

**Affected users and/or printers**
All I believe.

**Additional context**
I've used Cura towers before with success and the tower provided in the model above off of thingiverse kept popping off during my prints. I wanted to be able to try using a cura generated tower instead, but I couldn't manage it without getting supports on a bunch of the rest of the model that didn't need it.

Ghostkeeper, 2020-01-27:
To think about why we'd need this, I'd like to come up with a logical predicate of which cases you want to have a tower but not bigger pieces of support. Because if you take any normal model you'd want to have the big areas supported more than the small areas.

I think your case is specific to this octopus model, where support would be placed in between the links of the tentacles. This support would be hard to remove. You can filter that out with X/Y distances. So maybe what you're asking should really be a method to not place support in places where it's hard to remove?

sunlanterns, 2020-01-27:
@Ghostkeeper 
Is there a way to filter out on X/Y already? I'm new enough to 3D printing I could have easily over-looked a setting because I didn't understand the label for it.

For the sake of flexibility it makes more sense to me to just allow for towers separate from supports. But if it's a difficult task for the engine, than I understand wanting to find an easier way to get the same conclusion. Having an x/y filter on both the min and max side of things would probably cover most situations like this. 

"Hard to Remove" would certainly be a cool feature, but that seems like it would be even harder to code, because programmatically determining what makes a support hard to remove doesn't seem straightforward to me. But like I said I'm still new-ish to 3D printing so it might be obvious to you all!

Ghostkeeper, 2020-01-29:
> Is there a way to filter out on X/Y already? I'm new enough to 3D printing I could have easily over-looked a setting because I didn't understand the label for it.

What I meant is by adjusting the [Support X/Y Distance](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_xy_distance.md) and [Minimum X/Y Distance](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_xy_distance_overhang.md) settings. Increasing these will force support to keep more distance from the model and thus won't let it get (as) stuck inside those tentacles.

> For the sake of flexibility it makes more sense to me to just allow for towers separate from supports. But if it's a difficult task for the engine, than I understand wanting to find an easier way to [...]

It's not all that difficult to program I think. But we also have to keep usability in mind. If nobody is going to use it and it's just confusing we're better off not adding it. So that's why I wanted to go deeper into what "situations like this" are. That way we can build something that's really effective against situations like this rather than another settings that people would just have to try out.