Ultimaker/Cura #7025: [4.4.1] Straight line curving at 90° intersection

Liger0, 2020-01-25:
**Application version**
4.4.1

**Platform**
Win10 x64

**Printer**
Cr10s

**Reproduction steps**
1. Use a model with lines that intersect at 90°

**Screenshot(s)**
[image]
[image]

**Actual results**
Depending on the intersection feature type, the lines curve at the end

**Expected results**
End the line straight

**Project file**
[straight line deviation.zip](https://github.com/Ultimaker/Cura/files/4112945/straight.line.deviation.zip)

smartavionics, 2020-01-25:
Use my Cura, it does thin walls differently to UM Cura...

[image]
[image]

Ghostkeeper, 2020-01-27:
You can try shifting or rotating the model slightly as well. This is the result of the technique used to fill small gaps and print walls thinner than the line width. It fills the thin space with tiny lines and then combines those lines together. The line segment at the end has an endpoint that's not connected to anything else and then doesn't get "straightened" to another line segment, so it gets a little lip there. I think something similar happens at the intersection since there is a bit more space there. By shifting or rotating the model slightly, the lines are generated relatively at a different offset and it could combine differently.

Ultimately the thin wall printing algorithm is something we need to look at (mostly because of the bug that it doesn't combine the lines well sometimes).

mahtDFR, 2020-01-29:
Thanks for the report, this is an issue we're aware of. It's a sub issue to a larger issue to do with a library and it's being looked into. We'll create a ticket in our internal system for this.

closed 2024-12-07 by github-actions[bot]