Ultimaker/Cura #7026: I can’t assign a second extruder to the second model.

125401qwerty, 2020-01-26:
Hi.

I need to print two models with different materials (PLA and ABS). 1 extruder PLA, 2 - ABS.

Problem: I can’t assign a second extruder to the second model. After rendering layers, both models are always assigned to the first extruder for printing.

How to solve this problem?

Cura Version 4.4.1.

In the 2nd and 3rd versions of Cura, this worked great. But I need this feature in version 4.

fieldOfView, 2020-01-26:
Please use the issue template. One of the items in the issue template is "steps to reproduce". This would be something like so:
*Steps to reproduce*
* Load first model into Cura
* Load second model into Cura
* Select second model and use the toolbar on the left to select the 2nd extruder
* Slice and inspect in Layer view

*Expected result*
Second model is shown to be printed with extruder 2

*Actual result*
Layerview shows both models will be printed with extruder 1

Did I get that correctly, or do you use other steps?

125401qwerty, 2020-01-26:
Yes. You got it right. Is this a bug or am I doing something wrong?

Ghostkeeper, 2020-01-27:
I'm not able to reproduce the problem. For me it seems to properly print the second object with the second extruder.

Here's a project file: [two_cubes_two_extruders.zip](https://github.com/Ultimaker/Cura/files/4117177/two_cubes_two_extruders.zip)
The second cube is printed with the second extruder, and in layer view if you set it to visualise the material colour you'll see that it is printed in green (ABS).

125401qwerty, 2020-01-27:
ok . I will try it.

closed 2020-06-19 by no-response[bot]