Ultimaker/Cura #7028: Want Infill Patterns to be exported as STL object. As it is already in algorithm because it is shoeing in G-Code.

devsid-99, 2020-01-27:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-01-27:
You'll need to populate the issue template with relevant info before we are able to help you effectively.

f you want to export an STL this is a CAD issue, as Cura will not do STL exports, it only uses STL as an input format.

From the title of the issue it seems like you want to create an object that is just infill and no skin. As a workaround, you can import an STL into Cura and set all skin/wall values to 0. No idea if I have understood you correctly so i suggest filling out the issue template completely.

Ghostkeeper, 2020-01-27:
From the title it seems he wants to somehow export the tubes that we show in layer view as STL files. This is not information we can export since it's generated in the GPU as a geometry shader. It never becomes known to the CPU at the moment. The CPU would have to reconstruct it.

That reconstruction can be done. We had a research project at some point that did this. The code is still in a branch here: https://github.com/Ultimaker/Cura/commit/bdbc32e915126c17ae29d0d388d0a32774e160a4

closed 2021-06-01 by no-response[bot]