Ultimaker/Cura #7031: New Feature request

jldavis1969, 2020-01-28:
As a user of your product “Ultimaker Cura”. I would like to thank you for such a wonderful product.
 
I would also like to see an addition to Cura in the form of a centering dot on the build plate image.
 
It should be an easy add, as all you would have to do is half the X, Y coordinates of the build plate that is referenced in the machine settings.
 
Thank you once again
Jeffrey Davis

mahtDFR, 2020-01-28:
Thanks for the feature request. Can you open it again as a feature request rather than an issue please? We have a [specific tempate to fill](https://github.com/Ultimaker/Cura/issues/new?assignees=&labels=Type%3A+New+Feature&template=feature_request.md&title=) for this.

closed 2020-01-28 by mahtDFR

fieldOfView, 2020-01-28:
Thanks for the kind words. You forgot to fill in the feature template. It may sound pedantic requiring that, but the Cura team is always overloaded with, so they need to know why your feature is more important than the other things they are working on.

You need to make a case why this feature is important, not just for you but for a large group of users. What can't you do without this feature? What sort of users suffer from not having this feature?

Also try to use a descriptive issue title next time you open a feature request. WIth your title of "New feature request", you always have to read the whole post to know exactly what the feature request is. Many developers receive updates on issues via email, and when they only see "Re: New Feature request", they have to go look for what the actual request was in previous message. "Centering dot on build plate" would have been a nice issue subject.