Ultimaker/Cura #7033: Problem using the spaghetti infill due to the temperature

Tcadiou, 2020-01-28:
**Application version**
CURA 4.3.0

**Platform**
Windows10 

**Printer**
Custom 3D printer

**Reproduction steps**
1. I used the "spaghetti infill" in the "Experimental" 
2. I have set the spaghetti's infill every few centimeters
3. I set the infill with a second extruder (the second extruder using paste)
4. Even if the the system should use the second head every few centimeters, it change the tool every time the lauer change

**Screenshot(s)**
Here is an example of the slicing with the methods:
[image]

Here is the gcode associated:
for a random layer that does not recquire the second extruder:
;LAYER:157
G92 E0
G1 X215 F3000
T0
G92 E0
M104 T1 S-24

for the next layer that still does not recquires the second extruder:
;LAYER:158
*... few lines*
T1
G92 E0
M105
M109 S-24
M107
M104 S-14
M204 S5000

**Actual results**
every times the second extruder is called, it changes on the printer even if the second extruder doesn't have to move
if I try to change the material settings, Cura crashes

**Expected results**
I want the second extruder to be called only when it has to extrude
I don't want it to have those "temperature check" 

**Project file**
[equerre non chanfrein.txt](https://github.com/Ultimaker/Cura/files/4121487/equerre.non.chanfrein.txt)

**Log file**
no log file

**Additional information**
no additionnal informations

mahtDFR, 2020-01-28:
Thanks for providing the gcode. 
[image]

I set up a custom FFF machine profile (default settings and 2 extruders, so there are likely some differences from your setup) and isolated the E2 movement and this is the result. If you can, share a .zip of a 3mf file so I am able to replicate your exact setup.

As I understand the issue, the extruder is switching to extuder 2 (T1) unecessarily, and m109 commands are wasting time waiting for extruders to change temperature when not needed. Have I understood that right?

As for the crash, are you able to share a log file as well? This will help to diagnose the exact cause of it.

Tcadiou, 2020-01-28:
Hi @mahtDFR ,
Thank you a lot
Here is the link of what I think is the log and a .3mf file that reproduce the same scheme and other infos?:
https://we.tl/t-F24AtWJQXS

I would have like to provide you the screenshots of the settings I have put for the material, but this is when it crashes.

You are right, they switch the extruder unnecesseraly because the extruder shouldn't be called at all for the unnecessarly layers.

mahtDFR, 2020-01-28:
The project file isn't importing the correct printer for me, probably because we're working with different versions. Another thought: it's possible that this has been fixed already in the latest version, so could you also do a version update to the latest version and let me know if the issue persists?

mahtDFR, 2020-01-29:
Just had a chat with the dev team about this. Since spaghetti infill is an experimental setting it's not really something we will actively support. The issue is already known to us, so thanks for the report either way. Unfortunately it's not something we can devote development time to right now, but I will defer this ticket in case anyone else can find a solution.

Tcadiou, 2020-01-29:
In any case, thank you a lot and sorry for the disturbance.
This is very kind of you and I will involve myself a bit more into that as well.
Thanks ;-)

Ghostkeeper, 2020-01-30:
I think this happens because CuraEngine determines which extruder will be used before it starts plotting the individual lines (only after the infill area is computed, not the individual lines yet). It doesn't realise then that most of the infill areas are going to become empty due to this spaghetti infill.

SamGoldFisher, 2020-08-18:
I tried again and faced the same problem. I forgot I created a post about it.
I seriously consider finding some time to work on that and other topic as well. Unfortunately I feel scared to contribute because I am a beginner. I am not confortable using git and working with other people (by lack of experience).
Is it just cloning locally the Cura repository, working on a topic and "pushing a work when it does work?
Any intel for a beginner would be appreciate. I would really like to contribute.

Ghostkeeper, 2020-08-18:
Well, Spaghetti Infill has been removed in Cura 4.7 anyway, so this won't get fixed, and you can't contribute any fixes to us about it any more. I'll close this issue. Sorry!

To contribute, the basic steps are:
* Create a Fork of the repository you want to make a change in. Cura has 3 main repositories: This one (`Ultimaker/Cura`) is the front-end. There is `Ultimaker/Uranium` for a front-end framework that Cura runs on. And then there is `Ultimaker/CuraEngine` for the slicing engine that generates g-code. For a change to the Spaghetti Infill, you'll be looking to fork CuraEngine. Creating a fork can be done on the Github website. You'll then have your own repository: `SamGoldFisher/CuraEngine`.
* Clone the fork to your local computer. Essentially this just downloads the source code to your computer and creates a Git directory for it there, so that you can make commits. The Github Desktop application is an easy (graphical) way to do this if you're not familiar with the Git commands or the terminal.
* Make changes to the source code.
* Test your changes. In the case of CuraEngine this involves compilation as well, which I'll not go into here. The wiki pages here on Github may help you out.
* Push your changes. They will be on your local fork then.
* Go back to Github.com and visit the original `Ultimaker/CuraEngine` and it'll show a pop-up message to automatically create a pull request with your changes.

But again, you won't be able to do this for a fix to Spaghetti Infill since the feature has been removed entirely.

closed 2020-08-18 by Ghostkeeper