Ultimaker/Cura #7036: Allow ironing to do the entire skin

Liger0, 2020-01-28:
**Is your feature request related to a problem? Please describe.**
If an area has to be printed on next layers, ironing will stop before.
This makes a not cool visual result around that between what is ironed and the unironed margin

[image]

**Describe the solution you'd like**
Insert an additional option that ignored the upper geometries

**Describe alternatives you've considered**
Make the ironing perfectly touch the part that have to be printed on next layer. It may make it harder to print on if the ironing isn't perfect in this case, so it isn't the best solution in my opinion

**Affected users and/or printers**
Cr10S, probably all.

Ghostkeeper, 2020-01-30:
This feature already exists. Set the [Ironing Inset](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/shell/ironing_inset.md) setting to 0 if you want it to match up perfectly.

Liger0, 2020-01-30:
It doesn't work. There is no difference no matter what number I use in the Ironing Insert.

Ghostkeeper, 2020-02-03:
You can see a difference in your screenshots. In the original screenshot you saw the zig-zaggy connections at the end of the lines. In the new screenshot you can't see them because they are *inside the much larger red line*. Try looking at the differences in a diff tool, or the final print result. It's ironing to the centreline of the outer wall if you set it to 0.

Liger0, 2020-02-03:
Yeah but I don't want to reach the outer part but the part where the upper cilinder is

Ghostkeeper, 2020-02-05:
Oh, then you want the setting to be different in different parts of your model. You'll need to add a modifier mesh to do that. Place a cube around the part where you want the offset to be 0 (the entire layer), and in the per-object settings tool, change that object to "modify settings for overlap". Then change the inset to 0 in that model's per-object settings.

Liger0, 2020-02-05:
Wouldn't it be possible to add an option not to consider the upper layers when considering the area inside the perimeters of the ironing?

Ghostkeeper, 2020-02-10:
I'll re-open this as a discussion.

Yes it's possible to add a setting to filter cases in which we can do the offset, but before that we need to consider whether this is a generic problem, the use case is big enough, and it can be explained. We can add settings all we like but if nobody is going to adjust them it just serves as clutter. For instance, what do you think the setting name would be?

Liger0, 2020-02-10:
Something like "Iron all the skin area", where the description would say "This option doesn't consider the geometries of the next layers and will iron on all the available surface defined inside the perimeter".
This not only would make the layer better not having no-ironed space between upper geometries and the bottom layer, but would avoid the crest to form in the "problematic area" where the next geometries will be generated, reducing the possibilities of failed prints, which I am seeing frequently.

Ghostkeeper, 2020-02-12:
It does need to consider the geometries of the next layers, though? You wouldn't want it to iron all of the 10 skin layers, so you only want it to iron up until where the next layer starts?

Liger0, 2020-02-12:
That's a possibility, maybe a setting to decide the maximum "hidden" area to iron would work, so parts with very big hidden are can be avoided while parts with little areas, like the last top rectancular layer before the cilinder is in my example, can be ironed totally avoiding ironing all around and leaving crests and still a bad circle all around.

printingotb, 2020-03-05:
I noticed a possible bug related to this. If I change top/bottom line width, the inset changes in relation to the outer edge of the outer wall (both on top of the of the higher geometry and around the base, see picture). This seems strange to me. Shouldn't the inset origin from the outer edge of the outer wall, regardless of line widths?

If it for some reason that I don't understand needs to be calculated from the top/bottom lines, shouldn't it be the top surface line width instead?

To continue Liger0's discussion, I think that the ironing lines going close to the base of higher geometry, should simply not be affected by the inset setting. Or maybe run to the inner edge of the outer wall to secure a full cover of the visible area.
It could possibly also use the inset but with a negative value instead.
Or will these suggestions make the different insets collide when higher geometry starts at an edge of the lower geometry, causing issues?

[image]

Ghostkeeper, 2020-03-09:
> I noticed a possible bug related to this. If I change top/bottom line width, the inset changes in relation to the outer edge of the outer wall (both on top of the of the higher geometry and around [...]

I guess that's because by default the inset is set to half a line width:

https://github.com/Ultimaker/Cura/blob/d041131f84ead913734edb3bb83b88175ebdcd67/resources/definitions/fdmprinter.def.json#L1586

Or are you seeing this even if the inset is hard-set to 0?

It's set to half a line width because if you're extruding all the way to the edge of where the model ends, you're going to extrude a little bit over the edge. If you do that you get an ugly serrated edge there. In theory setting it to half the outer wall line width will make the ironing lines stop in the thickest parts of the outer wall, then follow exactly the centreline of the outer wall before returning in the other direction. When I optimised the initial settings for ironing, this worked well to prevent the serrated edges.

printingotb, 2020-03-09:
I should perhaps have clarified that I was referring to the inset origin, not it's value.

I have all line-widths hard set to different values. But the same happens with all calculated.
No matter what line width I change, except for top/bottom, the inset origin keeps the same distance to the outer wall edge. When changing top/bottom width however, it is as if the inset origin moves and I need to adjust the inset value to get it to the same place (in relation to the edge of the outer wall). I suppose it believes that the outer wall width is the same as top/bottom? And the origin is half that width?

I have seen the serrated edges, I like to tune the inset to the smallest possible while still avoiding that. I'm at 0.12mm now but even that isn't enough to completely hide non-ironed areas at the base of higher geometry.
I believe that the best would be if the inset origin is always calculated from the outer edge of the outer wall, regardless of the outer wall line width (or other line widths). That way we can use different outer wall line widths (same nozzle) on different prints and not need to change the inset. As for the calculated inset value, could it be half the nozzle diameter instead of line width?

I attached a project 
(On a side note, I discovered that if I use a top skin removal width that is too wide (leaving gaps in the top layers), it still tries to iron the gap. Perhaps I don't need to use too much removal width though hehe.)
[IronInsetTest.zip](https://github.com/Ultimaker/Cura/files/4307110/IronInsetTest.zip)

Ghostkeeper, 2020-03-13:
I think I understand now what the confusion is about, from you and Liger0. It's about the ZigZag pattern. The ZigZag pattern stops half a line width before the border, because the g-code commands need to follow the centreline of the path. Half of the line falls between the centreline of the "zags" and the border, so the centreline keeps half a line width distance.

The line width of ironing is still just the skin line width. Only with Ironing you commonly have a very low flow rate (7.95% in Printingotb's project). With this low flow rate the lines in layer view are also drawn very thinly. You expect these thin lines to match up with the border, while CuraEngine tries to match the actual line width (100% flow) to the border. This can be seen by setting the pattern to "lines" (after a simple modification to fdmprinter.def.json). The endpoints of the lines match up with the border then because the lines pattern doesn't move alongside the walls. Also setting the inset in Printingotb's project to 0, I get this then:
[image]

Or zooming in on where walls rest on the ironed surface:
[image]

I'm thinking that this is all a misunderstanding due to how flow is displayed in layer view.

printingotb, 2020-03-14:
You are right, I printed some tests, 0.4mm wide lines with 0 or 0.2 inset and it works as expected. 0 inset gives serrated edges, 0.2 doesn't but it leaves a very thin un-ironed area closest to the higher geometry. Going down to 0.15 almost eliminates it completely.
It wouldn't hurt if the area at the base would extend into the base a bit though. But I can work with this.

If I understand this correctly, the ironing flow is a % of the flow for the line width? (more specific top/bottom line width).
So, if I keep the same ironing line spacing but increase the line width to say 5mm (to exaggerate it), I will get more material in the same space during Ironing?
[image]
I printed a test but I reduced iron flow to 0.64% to avoid massive over-extrusion (0.4mm is 8% of 5mm so I reduced iron flow to 8% of my initial value as well).
The result is a good looking ironed surface that covers everything except roughly 2.5mm from the edges.
[image]

My conclusion is that if I change top/bottom line width I need to adjust both ironing flow and inset to get the same ironing result. I often use extra wide top/bottom lines for time saving and a few top surface layers with normal width for looks. But perhaps I should stick to the same line width values to avoid having top re-calibrate the ironing settings.

I don't know if any of this could be improved relatively easy but if it can, here are my suggestions:
-Ignore any inset at the base of higher geometry.
-Use the nozzle diameter to determine the toolpath where the zigzags turn instead of line width.
-Use the nozzle diameter to determine how much material to extrude instead of line width.

Thank you for your patience and I understand if this wont be prioritized.
Ps. I just realized that I haven't taken layer height into consideration.

Ghostkeeper, 2020-03-18:
> If I understand this correctly, the ironing flow is a % of the flow for the line width? (more specific top/bottom line width).

Yes, correct.

> So, if I keep the same ironing line spacing but increase the line width to say 5mm (to exaggerate it), I will get more material in the same space during Ironing?

Yeah, you'd get overextrusion (although compensated for by the ironing flow rate perhaps). It would also keep 2.5mm distance from the edges when using ZigZag.

Given your observations and Liger0's initial bug report, I think we need to improve the UX a bit here. I think it's not behaving as users would expect when using ZigZag (but Concentric is okay).

Ghostkeeper, 2020-03-23:
Developers, see issue CURA-7318.

Ghostkeeper, 2020-05-18:
This issue is fixed here: https://github.com/Ultimaker/CuraEngine/pull/1251

It now uses the flow-compensated width of the line to create the ironing pattern.

closed 2020-05-18 by Ghostkeeper