Ultimaker/Cura #7040: [4.4.1] Concentric pattern is not so concentric

LightTemplar, 2020-01-29:
**Application version**
4.4.1

**Platform**
KDE Neon

**Printer**
Ender 5

**Reproduction steps**
1. loaded STL with a big round base.
2. Set "Bottom Pattern Initial Layer": Concentric
3. Sliced

**Screenshot(s)**
[image]
 

**Actual results**
All lines on the first layer are crooked.

**Expected results**
All lines on the first layer are circular or spiral

**Project file**
https://we.tl/t-87WDQ7MgTN

smartavionics, 2020-01-29:
I found that increasing the value of Maximum Resolution to 0.5 mostly fixed this issue. Not perfect but quite a bit better...

[image]

smartavionics, 2020-01-29:
I think I have a fix for Cura. With this fix and your project unchanged, it now looks like this...

[image]

closed 2020-02-05 by rburema