Ultimaker/Cura #7042: [4.4.1] Stringing and underextrusion when there are supports

Liger0, 2020-01-29:
**Application version**
4.4.1

**Platform**
Win10x64

**Printer**
Cr10S

**Reproduction steps**
1. Use supports
2. Enable combing (I tried All and In-infill)
3. Printer with Estep, flow and Marlin's Linear Advance perfectly tuned)

**Screenshot(s)**

Give me the time to do them, I'll update.

**Actual results**
As all is calibrated, a minimal material loss will make gaps in some other parts. 
I have no stringing on stringing tests.
Supports are partially missing and print's perimeters so thin that they don't touch because of the previously missed plastic.
All my prints fail if there is support.

**Expected results**
No filament (even solid parts like they were printed) hanging.

**Project file**

[ooze between supports profile.zip](https://github.com/Ultimaker/Cura/files/4131130/ooze.between.supports.profile.zip)

Gcode on request privately if needed

**Additional information**
I see both light blue and dark blue lines in the preview. I can't find which is which (retracted or un-retracted) in the legend, but there shouldn't be any un-retracted travel either way.

Liger0, 2020-01-30:
Oh well, I totally forgot that setting, I'm sorry.
I set up combing All with 1mm of max distance, then 50 retractions within 3mm and it now doesn't show dark blue lines.
Would it be possible to show the light and dark blue lines in the legend? Because it doesn't show the difference.

rburema, 2020-01-30:
> Oh well, I totally forgot that setting, I'm sorry.

We have a lot of settings, no need to be sorry :-)

> Would it be possible to show the light and dark blue lines in the legend? Because it doesn't show the difference.

I'll have a look/ask if we can spend some time on that.

Does it work for your actual print as well?

Liger0, 2020-01-30:
I'll let you know when I have to print something like that again, thanks.

Ghostkeeper, 2020-01-31:
A long time ago we had the idea to make the legend a bit more clear by drawing all applicable colours in the box. So the travel move would have two triangles in the legend colour box, one dark blue and one light blue. We never got around to it.

The original issue is found not to be a problem, so I'll close this issue then.

closed 2020-01-31 by Ghostkeeper