Ultimaker/Cura #7043: Cura[4.4.1]

SRaynor96, 2020-01-29:
Cura 4.4.1 is refusing to allow me to slice anything.

Thank you
Sebastian

rburema, 2020-01-30:
Does your firewall block ports on localhost?

mahtDFR, 2020-01-30:
Hi Sebastian, thanks for the report.

Can you please send log data or a project file? We need some information to be able to diagnose the issue, your statement above isn't a lot to go on. Please see the issue template below and fill in as much detail as possible:

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Liger0, 2020-01-30:
Most of the times I also am unable to slice.
I just have to mod a setting and write it back and it works again, until I modify something again.

mahtDFR, 2020-01-30:
Without some specific info, I am unable to help you. Please refer to the answer above and fill out the issue template in as much detail as you can.

Liger0, 2020-01-30:
For me it is:

OS: Win10x64
Printer: Cr10s
Log: 
[cura.zip](https://github.com/Ultimaker/Cura/files/4134797/cura.zip)

I'll check for projects when it happens again.

mahtDFR, 2020-01-31:
``2020-01-30 15:09:38,351 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [663]: Slicing took 6.988180875778198 seconds`` Hi @Liger0 I am seeing this in your log. I can't reproduce the issue without some more detail, so please provide any if you can.

Liger0, 2020-02-15:
Today it happened again (4.5 beta).
I'll post log and screenshot, for the project tell me where should I send it privately if needed.
[cura.zip](https://github.com/Ultimaker/Cura/files/4209053/cura.zip)
[image]

It seems random, this time it just sliced correctly but I was using supports by default and I forgot to disable them since I didn't need them, so I disabled them, and it failed slicing with them off.
I then enabled support and disabled it again (without slicing between the clicks), sliced again and this time it worked. It seems totally random.

Ghostkeeper, 2020-02-18:
I see this in the log:

```
2020-02-15 23:18:05,551 - DEBUG - [Thread-1] UM.FileHandler.WriteFileJob.run [76]: Writing file took 1.6675128936767578 seconds
2020-02-15 23:18:05,789 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [328]: request[07b9cf54][post][PyQt5.QtCore.QUrl('https://stats.ultimaker.com/api/cura')][timeout=None][b'{"machine_'...] got an error 6, SSL handshake failed
2020-02-15 23:18:05,793 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [349]: request[07b9cf54][post][PyQt5.QtCore.QUrl('https://stats.ultimaker.com/api/cura')][timeout=None][b'{"machine_'...] error callback scheduled
2020-02-15 23:18:05,797 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-02-15 23:18:05,805 - ERROR - [MainThread] SliceInfoPlugin.SliceInfo._onRequestError [282]: Got error for SliceInfo request: SSL handshake failed
```

It's failing to send the anonymous statistics to our stats server. Perhaps that is what prevented the slice? It shouldn't prevent the slice. That would explain why it's random though.

rburema, 2020-02-18:
Just discussed this with @Ghostkeeper and that (fail in anon. stats) is very unlikely to be the cause, since that only happens _after_ the slicing data has been written. So something else must be going on.

Liger0, 2020-03-12:
It happened again. This time even changing supports on and off multiple times didn't unlock it.

[cura.zip](https://github.com/Ultimaker/Cura/files/4326943/cura.zip)
[Slicing fails.zip](https://github.com/Ultimaker/Cura/files/4326944/Slicing.fails.zip)

rburema, 2020-03-17:
In those logs, I see this happen a number of times:

```
2020-03-12 23:41:56,629 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [225]: Starting to slice...
2020-03-12 23:41:56,630 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [243]: Going to slice build plate [0]!
2020-03-12 23:41:56,634 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [292]: Attempting to kill the engine process
2020-03-12 23:41:56,635 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [298]: Killing engine process
2020-03-12 23:41:56,639 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [301]: Engine process is killed. Received return code 1
2020-03-12 23:41:56,647 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [213]: Previous socket existed. Closing that first.
```

Does it still say 'unable to slice' when slicing doesn't happen? If not, it could be that the click is somehow registered multiple times.

Liger0, 2020-03-17:
Yes, it says "unable to slice"

closed 2020-06-19 by no-response[bot]