Ultimaker/Cura #7044: Layer Selector missing from 3.6 to 4.4.1

Stup1dTwitB0ts, 2020-01-30:
**Application version**
4.4.1

**Platform**
Win 10 Pro Version 1903

**Printer**
Alfawise U20 ONE

**Reproduction steps**
1. Was running Cura v3.6 - was able to see layer sliders and layer selector in preview mode
2. Installed Cura 4.4.0 - lost the layer sliders and selector
3. Installed Cura 4.4.1 - still not there
4. Back leveled to Cura v4.3 then 4,2 then 4.1 and finally 3.6 - layer selector on available.

**Screenshot(s)**

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
To see the layer selectors so I can determine a layer at which to change the filament

**Project file**
N/A

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4132220/cura.log)

**Additional information**
none

rburema, 2020-01-30:
The sliders have moved. If you go to the Preview stage, there should be the layer slider at the very right of the screen (past even the print settings), and one for the 'time within layer' in the middle at the very bottom of the screen.

Stup1dTwitB0ts, 2020-02-01:
rburema, thanks for the reply.

I still don't see them. 

[image]

smartavionics, 2020-02-01:
Have you actually sliced the model? What happens when you click the slice button hiding under the settings?

Stup1dTwitB0ts, 2020-02-01:
Well I'm a damned idiot.

Thank you @smartavionics!

Ghostkeeper, 2020-02-03:
Added a warning to let you know next time.

commit 2c28f700 (2020-02-03): Add warning if you haven't sliced before going to layer view

Done as a 5 minute fix.

Fixes #7044.
closed 2020-02-03 by Ghostkeeper