Ultimaker/Cura #7046: flashforge dreamer3d profile only works with 1 extruder. activating the 2nd one adds bad gcode

valueerrorx, 2020-01-30:
**Application version**
4.4.1

**Platform**
ubuntu 18.04

**Printer**
flashforge dreamer3d

**Reproduction steps**
activate the second extruder
slice !
cura will add a line with "T0" before the startcode which renders the file unusable

**Actual results**
cura will add a line with "T0" before the startcode which renders the file unusable
if you manually remove T0 the flashforge dreamer3d will start printing but it will print everything with the same extruder

**Expected results**
don't add unusable gcode to the file
print with both extruders when defined in cura

**Project file**
look at line 10!
[dual2.gcode.txt](https://github.com/Ultimaker/Cura/files/4133849/dual2.gcode.txt)

**Additional information**
i created a github repo with the flashforge dreamer3d cura profile

https://github.com/valueerrorx/Flashforge-Dreamer3d-Cura-Definitions

Ghostkeeper, 2020-02-03:
That `T0` is there as the starting extruder (most likely your platform adhesion extruder). If your printer was left at `T1`, it'd need to switch. If it's already at `T0` it shouldn't have any effect. Why would it render the file unusable? It's pretty standard g-code.

valueerrorx, 2020-02-03:
well it just does..  if i direct the flashforge dreamer3d to this file it just says:
"cancel print" 
and goes back to the menu..   
removing the line T0 solves this problem 

and it shouldn't be necessary to add something to my g startcode
```M108 T{initial_extruder_nr};```
does the change when the rest of the initialisation process is done..

the thing is..  cura added a lot of things right before my gcode at first.. then i kinda fixed that by adding more information to the definition file and to the startcode..

TO is the only thing that remains and i can't get rid of that..  
makerbot is one of those gcode flavours that autormatically adds gcode to the startcode..  but i don't know how to prevent this

valueerrorx, 2020-02-03:
if i remove my startcode cura adds all of these lines
´´´ ;Generated with Cura_SteamEngine 4.4.1
T0
M190 S50
M104 S200
M104 S200
M116  ´´´

it seems that the dreamer3d expects the machine code first and therefore has a problem with T0

instead of deleting T0 i can change it into M106 T0  and the dreamer accepts the file

mahtDFR, 2020-02-04:
It seems to be a firmware issue with your printer. We don't have that particular printer here for testing so can't reproduce the issue. We don't have the development resources to create a workaround for this right now.

A possible workaround is to use a 'search and replace' post processing script, but this might strip all the extruder switch values.

Since the bug is not ours, I will close this ticket.

closed 2020-02-04 by mahtDFR

valueerrorx, 2020-02-04:
oke.. i see..    it was possible to influence what cura adds before the gcode by writing the correct gcode into the startcode..  therefore i thought someone from the team would know what setting is missing and needs to be added to the startcode in order to prevent cura from adding T0

thx anyway

valueerrorx, 2020-02-04:
search and replace will do ;-)  with regular expressions this a nobrainer..

Ghostkeeper, 2020-02-10:
If you have the regex to fix this issue it could help future troubleshooters trying to fix this issue :)