Ultimaker/Cura #7050: [4.1.1] Interface layer support without the main support using tree supports

Liger0, 2020-01-31:
**Application version**
4.4.1

**Platform**
Win10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Load project
2. Tree supports
3. 0.8mm xy distance

**Screenshot(s)**
[image]

**Actual results**
The interface layer dot is printed in the air even though nothing stands beneath them.
If I decrease x-y distance to 0.3, the tree are generated and the dot (dark blue) is is separated by 1 layer from the last tree layer (light blue)

**Expected results**
Only print interface layer dots if a tree is generated

**Project file**
[Dots in air.zip](https://github.com/Ultimaker/Cura/files/4140414/Dots.in.air.zip)

Liger0, 2020-02-01:
I am in general having issues with tree supports now because the latest layers are often miss even if the part is far from the model on x-y in the case that the part that gets supported is small.

I don't get why this happens, the fact that the model has a very little point as a overhang doesn't mean the support also does have to be that small, so small that the layers are not even generated.

luxology, 2020-02-03:
Use Support Blocker cube (E) on this dots.

Liger0, 2020-02-03:
Yeah but they are really a lot and it's a good waste of time.

mahtDFR, 2020-02-05:
Thanks for the report. I reproduced it and it's on our backog - devs: CURA-7168

nallath, 2020-07-13:
This has been fixed for 4.7

closed 2020-07-13 by nallath

Liger0, 2020-10-06:
Not solved. Recreated in #8477