Ultimaker/Cura #7051: [4.4.1] AppImage not starting up

MarekPikula, 2020-01-31:
**Application version**
4.4.1

**Platform**
X.org 1.20.6
Linux 5.4.13-201.fc31.x86_64 #1 SMP Tue Jan 21 17:21:47 UTC 2020 x86_64 GNU/Linux
Intel(R) Core(TM) i7-2620M CPU @ 2.70GHz with integrated graphics

**Printer**
Not relevant

**Reproduction steps**
1. Download Cura 4.4.1 AppImage from website
2. `chmod +x`
3. Run from console

**Screenshot(s)**
Not relevant

**Actual results**
Cura crashes

**Expected results**
Cura starts up

**Project file**
Not relevant

**Log file**
```
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Reinstalling the application may fix this problem.
/tmp/.mount_Cura 4cVipA1/usr/bin//cura.sh: linia 14: 116687 Przerwane               (zrzut pamięci) cura "$@"
```

**Additional information**
That's all

smartavionics, 2020-01-31:
If you run the AppImage from the console with the --debug option, what do you see on the console?

MarekPikula, 2020-01-31:
```
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Reinstalling the application may fix this problem.
Fatal Python error: Aborted

Current thread 0x00007f149bbde740 (most recent call first):
  File "/build/4.4/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 94 in __init__
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 176 in __init__
  File "/build/4.4/build/inst/bin/cura", line 179 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.4/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.4/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
/tmp/.mount_Cura 48LuWAj/usr/bin//cura.sh: linia 14: 136727 Przerwane               (zrzut pamięci) cura "$@"
```

smartavionics, 2020-01-31:
Not much more help. So, it can't load the xcb platform plugin. That could be due to an expected dynamic library cannot be found. The AppImage contains some dynamic libraries but it also expects some system libraries to be installed. Presumably, other X programs run with no problems?

smartavionics, 2020-01-31:
This is what I get on my Ubuntu Linux when I run one of my Cura releases, the libraries it uses could be different to the libraries used in the UM releases but should be fairly similar.
```
lsof -p 14252 | grep xcb
cura    14252 markb  mem       REG               0,51   102384 2929263557 /tmp/.mount_Cura-m0VHLB8/usr/bin/qt/plugins/xcbglintegrations/libqxcb-glx-integration.so
cura    14252 markb  mem       REG                8,2   137728   14039699 /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
cura    14252 markb  mem       REG                8,2    18560   14039671 /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0.0.0
cura    14252 markb  mem       REG                8,2   100480   14039675 /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0.0.0
cura    14252 markb  mem       REG                8,2     5976   14037027 /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1.0.0
cura    14252 markb  mem       REG                8,2    26784   14039691 /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1.0.0
cura    14252 markb  mem       REG                8,2    10368   14039680 /usr/lib/x86_64-linux-gnu/libxcb-present.so.0.0.0
cura    14252 markb  mem       REG                8,2    10384   14039673 /usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0.0.0
cura    14252 markb  mem       REG               0,51    27176 2929262709 /tmp/.mount_Cura-m0VHLB8/usr/bin/qt/plugins/platforms/libqxcb.so
```

MarekPikula, 2020-02-01:
Everything runs no problem. In fact Cura 4.1.0 runs just fine on this computer. All libraries you mentioned are present in `/usr/lib64/`.

I have another computer with KDE flavour of Fedora (the one experiencing the issue is Gnome) with exact same kernel and Xorg, but Nvidia GPU and it runs just fine 4.4.1 version. It loads correctly all required libraries.

I'll try to check in which version it breaks.

MarekPikula, 2020-02-01:
All right. I pinned down a problem.

I forgot to mention that I renamed executable to `Cura 4.4.1.AppImage` after download. You might think that it's not and issue. That's what I thought at first as well. I used this naming scheme (so with space) up to 4.1.0 on this computer and it seems to break from 4.2.1 on. So right after download, when its name doesn't contain space it works just fine. When I introduce space in the file name it crashes.

smartavionics, 2020-02-01:
Hi, thanks for that info. Yes, I can confirm that a space in the AppImage file name causes problems. I have tested a fix here and will submit that to the Cura devs.

nallath, 2020-02-04:
Thanks for the detailed investigation and the fix!

closed 2020-02-04 by nallath