Ultimaker/Cura #7052: I guess this is more like a feature request than a bug fix.  When printing a raft support, there are some high speed (non-printing) moves which are done at ths same z setting as the printing.  Often, the print head catches on some of the previously printes raft parts.  This usually results in a small gash in the printed area, but sometimes catches enough of the print to release it from the bed.  A small z hop would be nice to make printing the adhesion raft more certain.  I am considering writing a post processor to add the z hop for G0 commands on the raft layers.

streckerja, 2020-01-31:
@**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

fieldOfView, 2020-02-01:
Please use the issue template and don’t put the whole thing in the issue subject

streckerja, 2020-02-01:
Sorry

Ghostkeeper, 2020-02-04:
A post-processing script is not really the way to go for this sort of thing. We'd rather implement it in the engine.

However I can't reproduce your problem. For me it performs Z hops fine in the raft. However since Cura's normal layer view doesn't show Z hops, you need to save the g-code and load it back in, in order to see them:
[image]

In order to make sure Z hops are applied for everything in your raft, you need to:
* Make sure Z Hop When Retracted is enabled. Otherwise it'll not hop at all.
* Make sure Z Hop Only Over Printed Parts is disabled. Otherwise it'll only hop when it travels over a different part, which is pretty much never for a big rounded shape like the raft.
* Make sure Combing Mode is set to "off". Otherwise it'll just travel through the inside instead of retracting.

GregValiant, 2024-11-25:
I'm cleaning house.
Is this still a relevant request?  Can it be closed?

closed 2024-12-09 by github-actions[bot]