Ultimaker/Cura #7053: Can't change value

macaron16, 2020-02-01:
**Application version**
Cura 4.4.1

**Platform**
OsX Catalina on iMac

**Printer**
Geeetech A30M

**Reproduction steps**
change a value (calibration length for example)
validate it by clicking enter
slide
print
try to change the value by clicking the 'box'
no selection text' I ' appear; you can just double click but can't change the value
like if keyboard is 'lock'

need to restart Cura, to make it works again.

PS : sometime it's working; i don't find why or which exact operation lock the keyboard.

**Screenshot(s)**
no need

**Actual results**

**Expected results**
change the value ;-)

**Project file**

**Log file**

**Additional information**

fieldOfView, 2020-02-01:
Do you have the sidebar gui plugin installed? See https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/2

macaron16, 2020-02-01:
Yes. I forget it.
SideBar GUI v 4.0.6 is install.

I'll try to uninstall it.

Ghostkeeper, 2020-02-04:
I'm not able to reproduce this problem. My steps were:

1. Load a model.
2. Change wall thickness to 2 (up from 1.05).
3. Let it slice.
4. Send the print to the printer via wifi.
5. Change the wall thickness to 3. Just lets me change it.

I haven't seen an issue like this recently in my (not very) normal Cura usage.

fieldOfView, 2020-02-05:
This sounds like the issue in the Sidebar GUI plugin I linked to above. That issue now has a possible fix and a development snapshot to try: https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/2#issuecomment-582637223

closed 2020-06-19 by no-response[bot]