Ultimaker/Cura #7054: [4.4.1] Cooling fan behavior is flawed when using MakerBot G-Code flavor

eugr, 2020-02-01:
**Application version**
4.4.1

**Platform**
Platform-independent

**Printer**
Custom profile for FlashForge Creator Pro - see my repo at https://github.com/eugr/Flashforge-for-Cura, but can be reproduced with any Makerbot-style printer

**Reproduction steps**
1. Load STL
2. Set "Regular fan speed at layer" to 4
3. Slice/save to G-Code
4. Send to printer

**Actual results**
The cooling fan starts after the first layer

**Expected results**
The cooling fan starts after the third layer as FFCP has non-PWM fan (so it's either on or off). Maybe we could introduce a setting in printer definition that controls whether the printer has PWM or not?

**Additional information**
After analyzing the G-Code file, looks like Cura inserts M126 T0 at the start of each layer. For RepRap-style that would be M106 Sxxx, but MakerBot flavor M126 does not support speeds, so the result is that the speed is set to 100% right away.

Example:

```
;LAYER_COUNT:711
;LAYER:0
M127 T0
G0 F1800 X-25.477 Y-20.702 Z0.3
;TYPE:SKIRT
G1 F1800 E1.3
...... skip ....
;TIME_ELAPSED:158.450742
;LAYER:1
M140 S65
M126 T0 ; here it was supposed to gradually increase but starts spinning at 100% because MakerBot
...... skip ....
;LAYER:2
M126 T0 ; again
...... skip .....
;LAYER:3
M126 T0 ; that's where it was supposed to reach 100%
.... skip ....
;LAYER:4
;TYPE:WALL-INNER --- finally, no M126
```

Ghostkeeper, 2020-02-04:
Indeed I see that behaviour here:

https://github.com/Ultimaker/CuraEngine/blob/c375fd44774f21d73c620908ceffe3ac22c69630/src/gcodeExport.cpp#L1148-L1163

It's enabling the fan when the speed > 0%. You're saying that it should enable the fan when the speed > 50%?

eugr, 2020-02-04:
Well, I'd not enable a fan at all for Makerbot-style printers until you reach "regular" fan speed. This will make it more predictable when it turns on (it will turn on when regular speed at layer/height is reached). Something like that (assuming we don't get into some weird situation with floating point numbers when it never really reaches cool_fan_speed_min value). The value itself is "regular fan speed" from settings:

```
if (flavor == EGCodeFlavor::MAKERBOT && (speed >= cool_fan_speed_min) ) 
            *output_stream << "M126 T0" << new_line; //value = speed * 255 / 100 // Makerbot cannot set fan speed...;
        else 
```

But I think ideally there should also be a flag enabling or disabling gradual fan ramp up, so non-Makerbot users could also have a binary control over fan (off before, on after certain height/level).

Ghostkeeper, 2020-02-07:
> But I think ideally there should also be a flag enabling or disabling gradual fan ramp up, so non-Makerbot users could also have a binary control over fan (off before, on after certain height/level).

In general I'd prefer to get away from the whole notion of g-code flavours and toggle between individual differences, such as with firmware retract. It's not a change I can make now though.

We've decided on setting the drop-off point at 50% in the end still, rather than 100%. This is because many profiles (in particular material profiles) set the fan speed at 70% or 30% or so and then won't get fan speed even though they do normally have 70% fan speed. This way the fan speed will be more accurate for high temperature materials and still accurate for lower temperature materials. And your individual use case with the initial layer fan speed is also still possible, albeit with some weirdness. You can work around that by setting the regular fan speed to 50%.

closed 2020-02-07 by Ghostkeeper

eugr, 2020-02-12:
@Ghostkeeper - Thanks! Is it a part of 4.5 beta? I've just tried it and I'm seeing the same behavior as before...

fieldOfView, 2020-02-12:
No, the above fix is not yet part of Cura 4.5 but will be in Cura 4.6

Ghostkeeper, 2020-02-13:
Yeah, sorry, it was too late for this to be in the beta. We've split that off to get the 4.5 release in a more stable condition, otherwise there is not enough time to test everything.

eugr, 2020-02-14:
No problem, it can wait until the next release....

Meins321, 2021-07-14:
Yeah thank you for bringing this up!

The flashforge machines need some love with cura and your profile works and should be included with cura 4.11 and newer :-P