Ultimaker/Cura #7055: [4.4.1] Initial Layer Height gives wrong order

bakgaard, 2020-02-02:
**Application version**
4.4.1

**Platform**

* Windows 10, 1809 (OS build 17763.973)

**Printer**

* Local printer, Prusa MK2s.
* Setup is Heated bed, with Marlin G-code, one extruder, size 0.25 mm

**Reproduction steps**

* Load file with flat base, like a standard cylinder from Blender
  Both ``.stl``, ``.blend``, ``.gcode``, and ``.3mf`` can be [found here](https://drive.google.com/open?id=1xwl-wRTgIJzMHhO-2eJS_-Te4rzBhatU).

* Load it into Cura as ``.stl``.
* Activate ``Build Plate Adhesion`` set to ``Raft``.
* Set ``Quality``'s ``Initial Layer Height`` to the same value as the ``Layer Height``.
* Slice

So far so good.

* See screenshots - arrow indicates the first layer
* Layer 4: We start on top of the raft - no cylinder yet, as we want it to: perfect
* Layer 5: First layer put down: ok
* Layer 6: Put **below** Layer 5 - *not* ok
* Layer 7: Put in between Layer 5 and 6 - not ok either
* Layer 8: Put on the same layer as Layer 5 - That's a huge 'nope'!
* Layer 9: Put on top of both Layer 5 and 8 - better.

**Screenshot(s)**

[image]

**Actual results**

[image]

**Expected results**

I had *really* hoped, that each layer would *just* build on top of each other, but that is not the case.

**Project file**

[Found here](https://drive.google.com/open?id=1xwl-wRTgIJzMHhO-2eJS_-Te4rzBhatU).

**Log file**

Not required for this

**Additional information**

bakgaard, 2020-02-02:
Turns out there is a value in ``Build Plate Adhesion`` called ``Initial Layer Z Overlap``, which is set to three times the ``Layer Height``. This will shift everything down by that much... but the first layer.

bakgaard, 2020-02-02:
Update: There's a value in ``Build Plate Adhesion`` called ``Raft Air Gap`` which determines, and I quote:

> The gap between the *the final raft layer* and *the first layer* of the model. Only the first layer is raised by this amount to lower the bonding between the raft layer and the model.  Makes it easier to peel off the raft.

> **Affects**: Initial Layer Z Overlap

The problem is, that I have two entire layers below the first - Setting ``Initial Layer Z Overlap`` to ``0`` seems to solve my problem (currently testing), but I don't get why ``Initial Layer Z Overlap`` is bound to half the value to ``Raft Air Gap``.

Liger0, 2020-02-02:
Yeah I also noticed this behaviour with raft, with also the z gap not being respected. Never opened an issue because I didn't have the time to watch out more about it.

Ghostkeeper, 2020-02-04:
I think the Z overlap depends on the air gap because the bigger the air gap, the more the first layer sags down so the more you'd have to press on it to make the second layer stick. Something like that?

If it doesn't work properly for your printer, we can change the defaults for your printer specifically. But we don't have the MK2 in the mainline Cura release, so I can't change it for that one (without getting a whole new printer definition for that printer).

Liger0, 2020-02-04:
On cr10S, the gap between part and last raft layer is not respected, for example.

Ghostkeeper, 2020-02-10:
The air gap is respected (the gap between the raft and the first layer is exactly the Raft Air Gap), but if the Initial Layer Z Overlap is more than the initial layer, it results in the second layer ending up below the initial layer. From what I can see, CuraEngine is doing exactly what the settings ask of it. It respects the Raft Air Gap as well as the Z Overlap as best it can. But the settings are a little bit weird.

bakgaard, 2020-02-11:
> What should the default be, for which printer?

If the MK2 is not a definition, you can hardly change it.

But I would suggest the following settings all across the board:

* ``Raft Air Gap`` = ``Layer Height`` x 3
* ``Initial Layer Z Overlap`` = ``0``

Ghostkeeper, 2020-02-13:
Do those settings work better for your CR10S as well, Liger0? Indeed I can't change it for your MK2 printer since Cura doesn't have a definition for it, but if it's better for other printers we can change it there.

Liger0, 2020-02-13:
Well it depends on the layer height, for something like 0.4mm nozzle and 0.28LH, 3x layer height is very high. I even use 0 gap on those layer heights on s3d.

closed 2020-06-19 by no-response[bot]