Ultimaker/Cura #7056: Support for conditional G Code in start & end GCode

owendare, 2020-02-03:
**Is your feature request related to a problem? Please describe.**
Yes.
Reprap firmware now supports conditional G Code.  The syntax for the conditional statements requires indenting of the code to be executed.
At present, Cura strips leading spaces and tab indents from the start and finish G Code when the program closes.  This causes the conditional code to execute regardless of how the statement resolves.
eg
`if !move.axes[0].homed || !move.axes[1].homed || !move.axes[2].homed
       G28
// rest of start code`

becomes

`if !move.axes[0].homed || !move.axes[1].homed || !move.axes[2].homed
G28
// rest of start code`

causing the G28 too execute even if the system is already homed.

**Describe the solution you'd like**
Ability to define hard indents which are not stripped.
This could be done with the ascii code #9 or some other method.

**Describe alternatives you've considered**
It is possible to put the start and end GCode into a macro which can be called, however this will be problematic in that if Cura does not see certain values (temperature etc) in teh start code, it will automaticall add them which may circumvent the contents of the macro.

**Affected users and/or printers**
Affects anyone using reprap firmware >V3

**Additional context**
(Add any other context or screenshots about the feature request here.)

mahtDFR, 2020-02-04:
Sounds like a syntax checker. Not a huge benefit to us but shouldn't be too hard to implement. I will put it up for discussion and see what comes back.

mahtDFR, 2020-02-05:
After a quick discussion we think this is a good suggestion but just not something we have time for now, so I have had to defer it. We are open to pull requests that fix this issue.