Ultimaker/Cura #7057: Brim settings

DJDC13, 2020-02-03:
**Is your feature request related to a problem? Please describe.**
Sometimes the brim still prints on certain parts inside the print even when selecting the option to print outside the model.

**Describe the solution you'd like**
It would be really useful if the brim could be blocked in certain areas like the support blocker. This would help when printing moving parts that still need the edges held down by a brim. Like a hinge for example where you want the edges held down but you don't want the brim around the pin.

mahtDFR, 2020-02-04:
Thanks for the request. I've put it up for discussion and will let you know what comes of it.

Ghostkeeper, 2020-02-10:
After discussing, we've decided that this issue is a valid improvement we could build, but needs more development work than we find it worth, now. So we'll defer this for later.

DKXAM, 2023-01-16:
i'd still like to see this

cruse2382, 2023-07-11:
this would be a great addition, hope it gets done.

jordanlambrecht, 2023-12-23:
Hello from the future. This would still be an extremely useful feature to have. I couldn't get @Sekisback 's solution to work:

[image]
[image]
[image]

Sekisback, 2023-12-23:
> Hello from the future. This would still be an extremely useful feature to have. I couldn't get @Sekisback 's solution to work:
> <img alt="Screenshot 2023-12-23 at 01 32 21" width="666" src="https://private-user-images.githubusercontent.com/54866853/292614509-86386eb0-b9fc-4a36-86cc-c4b08e5f9823.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMzMjAzNDgsIm5iZiI6MTcwMzMyMDA0OCwicGF0aCI6Ii81NDg2Njg1My8yOTI2MTQ1MDktODYzODZlYjAtYjlmYy00YTM2LTg2Y2MtYzRiMDhlNWY5ODIzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjIzVDA4MjcyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFkN2M2YmZkMTY2ZDhkNjc3NDNiMmNhNDI4YmUyNWVhNzljODJhZGYyZDg4NTRiMTVhYjU1Mzg4NWNmNzIwMWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0cAQokg9fus1j5btS9aixb2RB1fEhALJU-I2KcO8LMw"> <img alt="Screenshot 2023-12-23 at 01 32 15" width="615" src="https://private-user-images.githubusercontent.com/54866853/292614512-cdb4ae2f-79f8-4367-9095-8252d3870857.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMzMjAzNDgsIm5iZiI6MTcwMzMyMDA0OCwicGF0aCI6Ii81NDg2Njg1My8yOTI2MTQ1MTItY2RiNGFlMmYtNzlmOC00MzY3LTkwOTUtODI1MmQzODcwODU3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjIzVDA4MjcyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM4ZGU0Y2MyMzE5NmIwZGE2MWY4YTY4ZDg3YzZjZTk4NDlmZTExMWMwNmQ4NjRkOWI4MDQyNzc2ZTBlNWJhMTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Osv59Ue-YzMANSjKmYWbjysoskg3QD1phwQENT3MYiU"> <img alt="Screenshot 2023-12-23 at 01 31 59" width="418" src="https://private-user-images.githubusercontent.com/54866853/292614513-6e9599df-2a7e-4ba4-a7ff-d93cedeb2ca3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMzMjAzNDgsIm5iZiI6MTcwMzMyMDA0OCwicGF0aCI6Ii81NDg2Njg1My8yOTI2MTQ1MTMtNmU5NTk5ZGYtMmE3ZS00YmE0LWE3ZmYtZDkzY2VkZWIyY2EzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjIzVDA4MjcyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY5NDI5NTQ4NmM5ZDdiOWQ3ZTE3MDlkNjI1ZGQ4NjZiZWNmM2YxMzk3YThhMWZiMzQxMzdhZWI0MDgwNDAyNTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.F7BbwEgHzc3fiHWOoIEgww3Ulksq_tDgm7F6FjM0usw">

in cura is a setting for brim only on outside, that should work like expected

closed 2025-01-05 by github-actions[bot]