Ultimaker/Cura #7060: [4.4.1] Unsupported points

Liger0, 2020-02-03:
**Application version**
4.4.1

**Platform**
Win10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Use supports on parts that start with little dots
2. Minimum support area: 0

**Screenshot(s)**
[image]
[image]

**Actual results**
No support

**Expected results**
Supported points

**Project file**
[unsupported.zip](https://github.com/Ultimaker/Cura/files/4150062/unsupported.zip)

mahtDFR, 2020-02-04:
Thanks for the report. I could reproduce the results and it's a bug. Also tried enabling 'use towers' which gave some weird results. It's up for discussion.

mahtDFR, 2020-02-05:
We will investigate. I've added this to our backlog - CURA-7167

Block137, 2020-02-08:
Might not be related but by setting "Support Join Distance" to higher number, some support structure got removed. Worth looking into. If it's not related I'll make another bug report.

mahtDFR, 2020-07-02:
We have decided not to prioritise this in our sprints.
For now, a workaround is to use Support Horizontal Expansion.

closed 2024-12-07 by github-actions[bot]