Ultimaker/Cura #7062: [4.4.1] Tool change T0 and T1 for temperature Sets

amcubed, 2020-02-04:
I have a 3d Printer with 2 extruders.
After i use this version, i got a problema with tools and temperatures control.

So the orders versions, the gcode was:
"M104 T1 S200
M109 T1 S200"

"M104 T0 S200
M109 T0 S200"

And now they are on that way:

"T1
M104 S200
M109 S200"

"T0
M104 S200
M109 S200"

So, every time i use Extruder "T0" , he work fine, but when i change to Extruder T1, they will heat the Extruder T1 and T0.

And i dont need that way.
How i can solve that problem?
Thanks

mahtDFR, 2020-02-04:
Please fill in the bug template.

amcubed, 2020-02-04:
Its a custom FFF Printer.
He as 2 extruders.
And i have done a new Custom FFF Printer with reprap g-code flavor.
[bug_tools_temps.zip](https://github.com/Ultimaker/Cura/files/4153439/bug_tools_temps.zip)

Thanks for you help!
Best Regards

mahtDFR, 2020-02-04:
I can't reproduce the issue with the file you've provided.

In 4.4.1 can you please go to File > Save... and save out the file. This will then contain your printer settings along with the model. You can then zip it and upload it here.

amcubed, 2020-02-04:
Check now please!
thanks
[bug_tools_temps.zip](https://github.com/Ultimaker/Cura/files/4153933/bug_tools_temps.zip)

amcubed, 2020-02-05:
@mahtDFR 
Any news?
Thansk

Ghostkeeper, 2020-02-06:
The project file you provided only uses one extruder, T1. In the g-code I only see one extruder switch: the one at the beginning to T1.
Here is the g-code that I got from your project file: [test.zip](https://github.com/Ultimaker/Cura/files/4163961/test.zip)

Seems fine to me, right? I don't see any unnecessary switches or heating up of unused extruders.

amcubed, 2020-02-06:
Yes! He calls T1, but the M104 and M109 doesnt call the Tool, so the command comes blank "M104 S200", he will heatup the T0, but if comes with "M104 T1 S200", he will heatup the T1 and not the T0.

Do you understand what i want to explain?
Cheers

Ghostkeeper, 2020-02-10:
Just to be clear, we're talking about this bit here:

```
;Generated with Cura_SteamEngine master
T1
M190 S60
M104 S200
M109 S200
```

As far as I know, if the `T` parameter is not included, it should set the temperature of the current extruder, not of a different extruder.

amcubed, 2020-02-10:
> Just to be clear, we're talking about this bit here:
> 
> ```
> ;Generated with Cura_SteamEngine master
> T1
> M190 S60
> M104 S200
> M109 S200
> ```
> 
> As far as I know, if the `T` parameter is not included, it should set the temperature of the current extruder, not of a different extruder.

@Ghostkeeper 

If i stay on that way, my 3d printer start both extruders temp.

Thats my problem.

If i use "M104 S200", my firmware assume that will be the T0 than the T1
If i use "M104 T1 S200" my firmware understand.

Thanks

Ghostkeeper, 2020-02-10:
So far, I'm leaning towards that your firmware is in the wrong here, not Cura. We've not encountered a g-code flavour that interprets it this way yet.

What firmware are you using?

closed 2020-06-19 by no-response[bot]