Ultimaker/Cura #7064: Ura automatically resizes objects

blair9095, 2020-02-04:
When using the “one at a time” feature when printing multiple objects at a time the objects are resized regardless of the number of objects (it resizes a single object) and regardless of object placement on print bed.  This is particularly annoying when printing multiple small objects that don’t even touch shaded areas.

Resizing occurs regardless of placement, number of objects, or size of objects.

nallath, 2020-02-04:
Resize them? The models are left intact. Do you perhaps mean the shadows around the model?

blair9095, 2020-02-04:
Yes, I misspoke.  The correct phrase is Build Volume Adjusted.

 

I’ve attached a screen shot to illustrate the issue.  The pieces are ~31mm and ~24mm. Both are round.  As you can see the bounding boxes don’t touch but the pieces, when printed do not match these dimensions.

 

F. Lawrance (Larry) Lenoff

(828) 767-1462

LLenoff@hotmail.com

nallath, 2020-02-05:
Screenshots don't work when you reply over email, sorry (it's a weird thing from github).

Anyhow, the changing of the shadows is completely intentional. Since you print them one at a time, Cura needs to take into account that the head might hit the other object.

Ghostkeeper, 2020-02-10:
This is intended behaviour. Like Nallath said, it's enlarging the collision areas to prevent the print head from hitting previously printed objects. So we'll close this issue then.

closed 2020-02-10 by Ghostkeeper

blair9095, 2020-02-10:
My issue with the resizing is it happens whether it is required or not.  I can place 2 x 4mm parts at opposite ends of the build area and they still are resized.  This is despite the distance between pieces substantially exceeding the size of the extruder package.

It seems the resizing occurs based on simply selecting “one at a time”.  Regardless of object size, placement or potential for collision.

Ghostkeeper, 2020-02-12:
Yes, but the disallowed areas are not actually being printed. They are larger, but it make any difference for your print unless you place them close enough that they intersect.

blair9095, 2020-02-12:
That is my point. They should only be changed if within boundary.   However they are changed regardless of position. It happened however I arranged. Close together, expected.  Far apart, not expected but occurs. 

Fredrick Lawrance Lenoff 
LLenoff@hotmail.com
(828) 767-1462

Ghostkeeper, 2020-02-14:
I don't think I'm following you properly. Can you provide clear reproduction steps? So far I'm following this:
> 1. Add two objects.
> 2. Set Print Sequence to one-at-a-time mode.
> 3. Arrange all objects.
> 
> Expected result: Objects are placed close together.
> Actual result: Objects are placed far apart.

When I try this, the objects are placed as close together as possible without the print head hitting the previously printed model:

[image]

It seems though that the visualisation of the print head shape is currently wrong. It's not doing the mirroring+intersection any more.