Ultimaker/Cura #7065: Support flow

maukcc, 2020-02-04:
**Describe the solution you'd like**

Is there a reason that Support flow is shared between extruders?

Liger0, 2020-02-04:
How could they be separate?
You can use an extruder for the perimeters/infill of the support and one for the interface layer, and use different flow on the feature

fieldOfView, 2020-02-04:
Yes. The reason is the supports are internally all calculated into a single mesh, and a single mesh has only one flow.

maukcc, 2020-02-05:
Thanks for the clarification

closed 2020-02-05 by maukcc

fieldOfView, 2020-02-05:
Another way of thinking about your question is that supports are only printed with one of the extruders. The value isn’t actually shared between extruders, but the value from the extruder that prints (all) the supports is used.

maukcc, 2020-02-05:
> the value from the extruder that prints (all) the supports is used.

this is actually not true, the lowest value is used

fieldOfView, 2020-02-05:
It should take the value of the Support Infill Extruder, per here:
https://github.com/Ultimaker/Cura/blob/4.4/resources/definitions/fdmprinter.def.json#L2590
If it doesn't, that's a bug.

maukcc, 2020-02-06:
It does as expected now